cncjs/cncjs #761: [feature request] GCODE filter / preprocessor

risacher, 2022-05-23:
#### Description

It would be helpful (to me, maybe others) to have the capability to run a filter / preprocessor against the GCODE before being executed.  

For example, my laser has a noticeable delay (200-300 ms) turning on after an M3 or M4 command, but it can change power levels instantaneously.   As a work-around, when using the laser, I replace all 'M5' commands with 'M3 S10'; which keeps it "hot" but reduces the power to a level that doesn't burn.     I've also used find&replace for other things like changing speeds & power without re-running the CAM software.

I envision this as maybe just applying a series of regular expressions to the GCODE?

#### Versions

- CNCjs: 1.9.25
- Node.js: 12.22.9
- NPM: 8.8.0

#### How Do You Install CNCjs?

- [x] NPM

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware
- [X] Desktop or Laptop

#### Operating System
- [X] Linux

missinglink, 2025-01-24:
This would also be useful to convert Z-axis commands to servo command for XY pen potters.

MitchBradley, 2025-01-24:
Some controller firmware, like FluidNC for example, has drivers for different "motor" types like RCServo.  Stuff like this can often be done better in firmware, because there are things that you cannot do, or that are very difficult to do, in Grbl GCode.

The opposite is also true - there are things like backlash compensation that can be done better in GCode, because you have more information about the high level milling strategy there.  For things like that, it would probably be better to have an offline GCode preprocessor instead of burying it in CNCjs.  An offline preprocessor would be able to handle files that are much larger than CNCjs can, would work with any UI program, and would be easier to maintain and enhance because it is isolated from the complex user interface code that is CNCjs's primary function.

Projects can get into trouble if they sprout too many features that are outside of their main focus.