cncjs/cncjs #762: Unwanted G10 L2 P1

AnTeBiOss, 2022-05-25:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.25
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

Hello,

Each time I perform a Z probe with the widget or G38.2 in console mode i got some unwanted G10 L2 P1 Z-68.82 that totally mess my Z0 

any idea about this ?

cheton, 2023-01-19:
May be you didn't correctly set the touch plate height?

https://github.com/cncjs/cncjs/blob/v1.9.28/src/app/widgets/Probe/index.jsx#L187-L214

```js
// Set the WCS 0 offset
gcode(`; Set the active WCS ${probeAxis}0`),
gcode('G10', {
  L: 20,
  P: mapWCSToP(wcs),
  [probeAxis]: touchPlateHeight
}),
```

closed 2023-01-21 by cheton

kolontsov, 2023-05-14:
Hi, had the same issue, found cncjs-pendant-gamepad to be a reason: https://github.com/balthisar/cncjs-pendant-gamepad/issues/11