cncjs/cncjs #773: $g loop?

otoyasan, 2022-08-03:
#### Description
hello.
After using other software for several years, I recently discovered CNCjs. I like it because it is very easy to see.
I have a question.
If you don't enter anything, it seems to automatically type a command like "$g" endlessly in a cycle of about 10 seconds like the image, can you stop it? I'm a slow manual typer, so this automatic cycle gets in the way a lot.
[image]

#### Versions
- CNCjs: 1.9.2.6

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2022-08-03:
Yes, cncjs issues $g every 10 seconds to determine Grbl's parser state.  That behavior is not configurable except by modifying the source code.  If someone wants to change it, the line of code that sets the $g interval is this line in src/controller/Grbl/GrblController.js:
```
    const toleranceTime = 10000; // 10 seconds
```

cheton, 2022-08-05:
It looks weired. I never see this kind of situation when connecting to my Grbl. Here is my screenshot:
[image]

May I know if your Grbl is a customized version? It seems there is little difference to the `$G` command:

Yours
[image]

Mine
[image]

otoyasan, 2022-08-13:
> Yes, cncjs issues $g every 10 seconds to determine Grbl's parser state. That behavior is not configurable except by modifying the source code. If someone wants to change it, the line of code that [...]

otoyasan, 2022-08-13:
@MitchBradley 
あなたのコメントに感謝します。さっそくsrc/controller/Grbl/GrblController.jsを探して60秒位にするつもりです。

otoyasan, 2022-08-13:
@cheton 
あなたの指摘で気づきました。確かにこれは奇妙です。
これは先月購入した3018proMAXに付属しているメインボードのGRBLです。1.1fです。
カスタマイズされているかは不明ですが、オフラインコントローラが使えるのでそれ用に何かは変えていると思われます。
メーカーに確認が必要なようですね。

cheton, 2022-08-13:
@otoyasan Your Grbl is probably a customized version made by China that produces an incompatible result of G-code parser state. In normal case, it will reply with a message starting with an `[GC:` indicator like below:

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#g---view-gcode-parser-state
```
[GC:G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 S0.0 F500.0]
```

The built-in parser (in CNCjs) requires a digit right after the M-code, or it will not parse correctly because a single M character seems invalid:
https://github.com/cncjs/cncjs/blob/master/src/server/controllers/Grbl/GrblLineParserResultParserState.js

Now you have two options:
1. Try upgrading your Grbl firmware to 1.1h, but you should keep previous settings before doing an upgrade.
2. [Find the source code](https://github.com/cncjs/cncjs/wiki/Troubleshooting#troubleshoot-electron-app-on-windows) in the downloaded app and modify the [regular expression](https://github.com/cncjs/cncjs/blob/v1.9.26/src/server/controllers/Grbl/GrblLineParserResultParserState.js#L18) to make it compatible with the proprietary Grbl controller.

Hope this may help.

otoyasan, 2022-08-16:
@cheton 
現在ショップと協議中です。まだサポートが有効なので、彼の対応を待ちたいと思います。
そういえばこのマシンを購入する時、grblのアップデート可能なボードなのか確認していませんでした。

otoyasan, 2022-08-18:
@cheton @MitchBradley 
ボードのサプライヤーは暗号処理していてファームウェアのフラッシュはできないといいました。
コンソールにGCが頻繁に出るのはcncjsの仕様ではありませんよね？多分。
candleで私のマシンを走らせると、visualizer画面の左上にあったGCが暫くするとresponseに移動して頻繁に表示され、双方の通信を妨害しているようでそのうちスピンドル以外はフリーズします。
販売店は既に私宛に新しいボードを送ってます。それが同じ症状出ないことを信じるのみです。

cheton, 2022-08-18:
@otoyasan CNCjs periodically sends `$G` to Grbl to query parser state every 0.5 seconds and tell Grbl to print all of the active gcode modes. When sending this command to Grbl, it will reply with a message starting with an `[GC:` indicator:

```
[GC:G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 S0.0 F500.0]
```

Regarding to your issue, CNCjs uses the regular expression below to parse parser state message. Unfortunately, this _RegExp_ requires an alpha character should be followed by at least one digit. 

```js
// https://github.com/cncjs/cncjs/blob/v1.9.26/src/server/controllers/Grbl/GrblLineParserResultParserState.js#L18
// * Grbl v0.9
//   [G38.2 G54 G17 G21 G91 G94 M0 M5 M9 T0 F20. S0.]
// * Grbl v1.1
//   [GC:G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 S0.0 F500.0]
static parse(line) {
    const r = line.match(/^\[(?:GC:)?((?:[a-zA-Z][0-9]+(?:\.[0-9]*)?\s*)+)\]$/);
```

For your case, I guess the vendor might NOT be able to fix this issue especially that they don't have the ability to flash Grbl, and I don't think CNCjs should make a patch release to support this kind of weird message format.

A quick workaround solution to it is to modify the code of your installed app, you can simply change `+` to `*` to accept a single `M` token.

```diff
- /^\[(?:GC:)?((?:[a-zA-Z][0-9]+(?:\.[0-9]*)?\s*)+)\]$/
+ /^\[(?:GC:)?((?:[a-zA-Z][0-9]*(?:\.[0-9]*)?\s*)+)\]$/
```

Typically you can find the code at `C:\Users\{YOUR_USER_NAME}\AppData\Local\Programs\cncjs-app\resources\app\server\controllers\Grbl\GrblLineParserResultParserState.js`. If you need further assistance, I can send you a modified file (`GrblLineParserResultParserState.js`) for the replacement.

otoyasan, 2022-08-19:
@cheton 
今日、新しいボードが届きました。 
[image]

 
全く問題ありません。コンソールの入力を邪魔されることもなくなりました。
ボード内のGRBLが壊れていたということがはっきりしました。元々解っていたことですが。
もしもまた同じ症状が出た時は、あなたのアドバイスに従うことにします。本当にありがとうございました。

closed 2022-08-19 by otoyasan