cncjs/cncjs #776: Ability to easily set an axis to half the current value

skasti, 2022-08-16:
#### Description

I often zero on a work piece by touching off on one side, zeroing the axis, then jogging to the opposite side and touching off there. Then I set the axis to half the displayed value. 

So for a 277.93 mm wide work piece and a 4 mm bit, I touch off and zero on the left side of the piece and jog to the right side and touch off there. the value then is 281.93 mm (the piece + bot). I then have to open my phone and divide this value by two and enter the result in the axis widget to zero the axis on the center of the piece.

Issues with this is
- Manual punching of values is error-prone
- Unnecessarily cumbersome process
- Often end up rounding, which causes inaccuracy when doing second op stuff
- Takes a lot of time

I would love if the axis widget just had a button to set the value of an axis to half its current value, this would make this process a lot faster and less error prone.

I might attempt solving this myself, but I thought I might make an issue anyway in case there actually is a feature I just haven't understood how to use or if someone have already done this but it isn't in the current release

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [ ] Download the CNCjs Desktop Application
- [x] Downloaded the source and built it

#### CNC Controller

- [x] Grbl (grblHAL)
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

MitchBradley, 2022-08-16:
The shopfloor-tablet UI has an RPN calculator on the axis entry buttons.  Touch the button and the calculator appears, with the current value entered.  Hit  `2 ÷ Goto` to accomplish your task

skasti, 2022-08-16:
@MitchBradley Hm, I took one look at the shopfloor-tablet UI and discarded it for being too simplistic 😅 I'll take a look, but my goal is still to have this functionality in the main axes widget :) I managed to build the source now, so at least headed the right direction. Been backend-exclusive for so long I have repressed any knowledge I had about developing frontend applications 🙈

Billiam, 2022-08-25:
Even if your touch offs are fully manual, I think you could handle this with a pair of macros.

The first would save the position of the first touch off, and the second would divide the difference between the old and new positions in half, and set that as zero.

See: New and Initial Tool for a similar process: https://github.com/cncjs/CNCjs-Macros/tree/master/Initial%20%26%20New%20Tool