cncjs/cncjs #782: Device Serial Port Disappearing when Cycling Power

lopezj1, 2022-10-01:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.27
- Node.js: 10.24.1
- NPM: 6.14.12

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

Raspberry Pi 3 Model B+
Raspbian GNU/Linux 10 (buster)
GRBL v1.1f
Carbide Motion Board 2.4e (Shapeoko 3)

I am having an issue when power cycling the power supply to the carbide motion board for the raspberry pi to detect the serial device over USB.

On a fresh reboot, ttyACM0 appears and I can open the connection and run the machine using CNCjs webpage. But as soon as I turn off the power supply to the carbide motion board and turn it back on, ttyACM0 no longer appears even upon refreshing the available ports.

This does not seem like normal behavior, since when connecting the carbide motion board to my Dell Laptop via USB with local desktop install of cncjs I can cycle power and still pickup the serial port when turning back on without having to reboot the PC.

I confirmed the following:
USB cable is good
Raspberry Pi sees the device (lsusb and dmesg | grep usb)

emcniece, 2022-10-01:
What OS is on your Dell laptop?

This behaviour is familiar to me on a few different unix systems. Does it help if you change USB ports or unlplug/replug the USB cable?

lopezj1, 2022-10-01:
My Dell is running Windows 10.  I have tried unplugging/replugging the USB cable into the same port on the pi and on the other 3 ports.  `dmesg | grep usb` shows that the device is connected and shows the manufacturer, vendor id, product id etc upon unplugging/replugging.  Running `ls /dev`, ttyACM0 port does not appear again and therefore doesn't show on cncjs on the dropdown of available ports.

MitchBradley, 2022-10-01:
It could be a problem in the udev rules that Linux uses to create /dev entries when devices appear.  Here is one related reference https://gist.github.com/edro15/1c6cd63894836ed982a7d88bef26e4af .  If this is hard to understand, I sympathize with you; udev is very complicated because it is trying to solve a very hard problem, namely trying to recognize an unbounded number of different kinds of devices that identify themselves in strange ways and change frequently.  You might have more luck asking on a Linux forum where there are lots more Linux experts than there are here.  At its core, this problem is unrelated to cncjs; cncjs just happens to be the program that is stumbling over it.  The question you really need to ask, in some forum where there are lots of topic experts, is "how do I make this particular USB serial device reappear under the correct /dev/* name when it is power-cycled".

closed 2022-10-01 by lopezj1