cncjs/cncjs #789: [Question] How do we send commands to CNCJS through socket.io without losing lines after a point?

frizensami, 2022-11-04:
#### Description

Hello, and thanks for making CNCJS. It's become the default interface for the machine we're using just for motion control (no cutting involved), and I'm currently writing a higher-level interface to it so that less-technical users can use our machine as well.

I am currently using the **Controller API** to send and receive data over websockets (following the wiki). Specifically, I do not want to load a gcode file ahead of time (as we are performing some actions dynamically on our end in between lines), so I am using the `command` socket event, with the `gcode` command string to send a single line of gcode at a time. This works for small tests, but I find that if we have to send a large number of commands, e.g, something like this:

```
    for (let i = 0; i < 50; i++) {
      const x = -((i * 25) % 400)
      const y = (i * 2) % 400
      // Our own controller that does socket.emit('command', 'gcode ...) 
      controller.sendGCode(`G0 X${x} Y${y}`)
    }
```
where I send 50 lines of GCode to the machine to produce a motion pattern, the problem is that after a certain line, the rest of the GCode does not get executed. I am using a GRBL controller, and I notice in CNCJS that the planner buffer and receive buffer both get to 0 (so I assume they are full). 

My question is: is this because CNCJS does not delay sending GCode when the GRBL planner/RX buffer is full? I'm sure it manages this somewhere since it is able to execute large static GCode files, many queued keyboard commands, etc, but does this not apply to gcode commands from the socket? 

Thank you very much in advance. 

 #### Versions

- CNCjs: 1.9.27
- Node.js: 18.4.0
- NPM: 8.12.1

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

MitchBradley, 2022-11-04:
You might have to study the code and put in some debugging to answer that question.  I don't know the answer off the top of my head.  Maybe someone else does, but I suspect that there are only a very few people in the world who have looked at that part of the code in detail.  You would need to look at src/server/controllers/Grbl/GrblController.js and src/server/lib/Feeder.js .  If the code path you are triggering goes through the feeder, then it might work.  If it bypasses the feeder, then it will probably jam up the Grbl buffer.  I suppose that another possibility is that feeder.next() is not getting called appropriately.  The flow of control in this area is extraordinarily complicated.

frizensami, 2022-11-05:
Thanks for the help @MitchBradley. I expect that other people did not encounter this because, as you mentioned, dynamically sending many lines of gcode is an uncommon use case. I see that other pendants just send the commands directly because they only send a few and are not likely to overflow the rx buffer:
- https://github.com/cncjs/cncjs-pendant-ps3/blob/master/index.js#L337
- https://github.com/cncjs/cncjs-shopfloor-tablet/blob/a44560bce2191dd087c2436b48a0d348ac7f34cc/src/app.js#L146

I managed a workaround by:
- Sending the **command** event as per normal, but only if we see enough space in the rx buffer (in the controller state). 
- Waiting for a new state to arrive from CNCJS through the **controller:state** event (which guarantees that we have an accurate value for the current state of the planner and rx buffer for the next time we send a command)
- then returning from the function

Thanks again!

closed 2022-11-05 by frizensami