cncjs/cncjs #794: Provide upstream curated `package-lock.json`

fooker, 2022-12-27:
#### Description

Currently, the repository does not provide a `package-lock.json`. Therefore there is no way to reproduce the exact dependency set as used to build the provided release.

Please consider providing the `package-lock.json` used to create release, by either make them part of the release source distribution or by manage it as part of the repository (the later is preferred).

This would also enable other benefits like caching of `node_modules` on CI by hashing the lock file.

#### Versions

- CNCjs: 1.9.x

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

emcniece, 2022-12-27:
Acknowledged! I'll see if I can dig a package-lock.json out of the CI runs for a starting reference.

william-orange, 2022-12-30:
I have been attempting to install CNCJS on a Raspberry PI without success.  There is a sea of NMP/Node.js version incompatibilities and warnings of missing packages using any of the installation instructions that I have found.  Would the json file resolve my difficulties?  It sounds like it might.

colin-campbell, 2023-01-14:
Best practice is as far as I know to commit yarn.lock or package-lock.json and run `yarn install --frozen-lockfile` or `npm ci` on the eh ... CI. Otherwise it's impossible to get a reproducible build with this stuff. At least that is what I beat my developers over the head with. 

However there is some voodoo going on with the build that _copies_ bits of package.json into another one in `src/` - unsure what all that's about.

emcniece, 2023-01-15:
Managed to get package-lock.json out of a CI build: https://gist.github.com/emcniece/a02e5f8b129a57109efc0e931167387d

I had to update the Appveyor/CI NodeJS version from 12 to 14 in order to get builds to pass. This might cause some distribution problems - the unit tests appear to pass, but OS integration will need manual confirmation.

If somebody wants to try building and running with this package-lock.json that would be a great help. I'll see if I can tag and trigger a build for testing.

colin-campbell, 2023-01-15:
One of the problems with package-lock.json is that `npm install` will update it :)
It's a natural part of distributed development that this file is updated by the people working on the project and committed together with their changes. 

CI though, shouldn't generally alter or produce a lock file in and of itself. 
   
Then we have this abomination. Which CI (or anyone/anything that actually packages a real release should produce). https://docs.npmjs.com/cli/v9/commands/npm-shrinkwrap

However if your top-level project is being used as a dependency of other projects, this file can give downstream users a real pain.

emcniece, 2023-01-15:
I have a sneaking suspicion that `package-lock.json` was gitignored because of the cross-platform bugs present in `npm` at the time (versions 5-7), eg. https://github.com/npm/npm/issues/16938. If this is true we should be able to upgrade Node and NPM versions, commit a consistent `package-lock.json`, and still have CI build/package/publish working apps.

Here's an in-progress PR to publish `npm-shrinkwrap.json` as a CI artifact: https://github.com/cncjs/cncjs/compare/master...emcniece:cncjs:fix/publish-package-lock?expand=1

If this works and the resulting file is the same across the 6 platform builds (https://ci.appveyor.com/project/emcniece/cncjs) then the next step might be to manually confirm that each of these builds results in functional apps in its respective platform.

cheton, 2023-01-19:
It's going to be resovled with the upcoming 1.10.x releases.

closed 2023-01-19 by cheton