cncjs/cncjs #795: How to properly restart a somehow stopped job??

iz3man, 2023-01-07:
This is not a cncjs specific "issue", and may somehow apply to other cnc programs as well, but I use cncjs, and look for a proper way to restart a broken job.

Eg. The grbl nc file created has some issues and triggers and end switch which makes the machine go into ALARM. Then you need to RESET and work coordinates are set to 0/0/0 again with the spindle at some completetly different location.
Or a mill breaks during operation and you need to restart it. or ....

How do I "save" the work coordinates? So I do a homing, which hopefully is precise enough to give me consistant 0/0/0 position of the mill. And then I move the spindle to the position I need and zero out in cncjs webfrontend and start the job.

I know could write down the spindle machine's coordinates and move to that position once the machine got and issue.

But isn't there a better way to do it?
I'm pretty sure this is just a complete noob's question and easy to be answered?!

Thanks!

MitchBradley, 2023-01-07:
I have never had much success in restarting interrupted jobs. The interruption itself is very likely to ruin the part.The one exception is when the job is an array of identical parts, so the ruined one can be discarded and the rest cut from the remaining stock. For that purpose, I have versions of my gcode program that do all 9 parts, only the last 8, only the last 7, etc.  An alternative is to rerun the entire job “air cutting” the ones that are already complete, but that requires precise rezeroing so you do not mar good parts that are still attached to the stock

iz3man, 2023-01-07:
Thanks Mitch. My main issue is 're-zeroing' as you call it. 

I just now had a broken end mill at the end of a 2 part's job. The machine returned to homing after the job was done. I replaced the mill and generated a new nc file which included only the cuts not done. That went very well as the machine still had homeing and work pieces coordinates intact.

The question is how to remove the "alarm state" w/o changing the work coordinates. Is that even possible?

MitchBradley, 2023-01-07:
If you are using Grbl and the alarm entry type requires a hard reset via Ctrl-X (that is the character that is sent to Grbl; I don't remember what button you have to press on CNCjs to send that and I do not have time to look it up), then Grbl is going to do whatever it does with a Ctrl-X reset, which might change the work coordinates.  I do not think that it changes them, but I am not sure.

MitchBradley, 2023-01-07:
When you said RESET, you did not specify how you did that.  Was it a power cycle of the controller, or did you press a physical reset button, or did you send Ctrl-X via some UI button?  If you want a good answer, you have to ask a precise question.

iz3man, 2023-01-07:
It seems RESET can be done in different ways, and it SEEMS outcome is different?!
I found the best way to use is RESET button in the WFE. 

[image]

But still this somehow alters the working coordinates, especially Z axis it seems.
Resetting directly on the machine is the worst as this doesn't seem to be communicated back to cncjs.

cheton, 2023-01-19:
It is recommended to run the HOMING operation before starting each new job to ensure a consistent starting position. 

Additionally, I remember that Snapmaker 2 has built-in support for resuming interrupted jobs due to power failure. 

I think @parachvte can help answer if the resuming part is easy to do on the software side.