cncjs/cncjs #802: Cannot find module '@babel/polyfill' with v1.10.0

Jakeler, 2023-01-23:
#### Description
After installing  the latest release via the [git method](https://github.com/cncjs/cncjs/wiki/Installation#git-installation) it fails to start.
Issue is the `require('@babel/polyfill');` in the [start script](https://github.com/cncjs/cncjs/blob/v1.10.0/bin/cncjs#L3) while the module was removed recently in 0469fe0118fa267550d38434a15be160e624996e from the package.json that is copied to `/dist/cncjs/`.
It only works when it is in the same tree as the main package.json/node_modules.

It seems to work fine on current nodejs if I simply drop the require line, but not sure if this is the best approach. Another would be to install the polyfill again or even a completely different installation method?

#### Versions

- CNCjs: 1.10.0
- Node.js: 14.21.2 - 19.4.0
- NPM: 8.19.2

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Windows
- [ ] Mac
- [x] Linux

cheton, 2023-01-23:
I apologize for the inconvenience caused by the issue in version 1.10.0. As a temporary solution, you can remove the line `require('@babel/polyfill')` from your startup script. This should resolve the problem without introducing new issues for Node.js 14.

Please note that I am currently on vacation for Chinese New Year and will not be able to release a patch until my return. I will release a new patch when back to work to address this issue in the next release.

Jakeler, 2023-01-23:
Thanks for the quick answer, fixed in the ArchLinux [package](https://aur.archlinux.org/cgit/aur.git/commit/?h=cncjs&id=bbb54dfb52a607b05e423c147bb4e0e37090aeb6) now.

closed 2023-01-26 by cheton