cncjs/cncjs #804: Issues running reboot command to raspberry pi

deependdesigns, 2023-01-28:
#### Description

Trying to execute a reboot through the web interface and I am seeing this in the logs:

Jan 27 19 cnc cncjs[516]: 2023-01-28T01:31:19.386Z - #033[32minfo#033[39m api:commands run: title="Reboot", commands="sudo /sbin/reboot"
Jan 27 19:31:19 cnc cncjs[516]: PID:877> sudo: unable to change to root gid: Operation not permitted
Jan 27 19:31:19 cnc cncjs[516]: PID:877> sudo: error initializing audit plugin sudoers_audit19:31:

#### Versions

- CNCjs: 1.10.1
- Node.js: 14.19.0
- NPM: 6.14.16

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2023-01-28:
To grant superuser privileges, you will need to add the user to the `/etc/sudoers` file. The instructions for doing so can be found in this article: https://raspberrypi.stackexchange.com/questions/7133/how-to-change-user-pi-sudo-permissions-how-to-add-other-accounts-with-different

closed 2023-01-29 by cheton

Ikswoknib, 2024-02-01:
Hi,
i am having the same Problem. 
Which user do i have to add to sudoers to get the shutdown command from cncjs to work?
User Pi is off course already in the sudoers list.