cncjs/cncjs #808: Startup settings for Tinyg g2

simonemarin, 2023-02-06:
#### Description

Is it possible to use a macro or some startup script to set the settings in Tinyg? 
Here are all the settings I need for my machine
{1tr:1.75}
{2tr:1.75}
{4tr:1.75}
{2po:1}
{4po:1}
{xvm:1200}
{xfr:1200}
{yvm:1200}
{yfr:1200}
tx

#### Versions

- CNCjs: 1.10.1

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ x] Windows
- [ ] Mac
- [ ] Linux

ril3y, 2023-02-06:
What board are you using?  You can just set a maco for this and send it.

simonemarin, 2023-02-06:
Arduino due and discrete stepper controllers, ok I see the Macro Widget I was trying with Commands. Tomorrow I will check with the machine at hand, thanks.

simonemarin, 2023-02-16:
Hallo I have updated the title as I was using Tinyg and not G2core

ril3y, 2023-02-17:
I am not sure you are using TinyG.  TinyG does not run on a Arduino Due.

simonemarin, 2023-02-17:
Yes it was 0.98 which is, if I am not mistaken, a Tinyg with G2core features (?)

ril3y, 2023-02-17:
TinyG does not run on a Arduino Due.  Its a stand alone board we sell.  https://synthetos.myshopify.com/products/tinyg Its the pre-cursor to g2core. 

G2core runs on a Arduino Due (among other boards we make). 

I just want to be very clear about this.  Do you have a v8 TinyG?  Or are you running g2core on a Arduino Due?

simonemarin, 2023-02-17:
Ok thank you for the clarification, I run it on a Due with discrete stepper drivers, directly wired into the Due.