cncjs/cncjs #811: Macos M1 damaged img

beckmx, 2023-02-14:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: cncjs-app-1.10.1-macos-arm64.dmg

#### How Do You Install CNCjs?

- [ X] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl

#### Hardware

- [ x] Macos M1 

#### Operating System

- [x ] Mac Ventura 13.1

There is a message that popsup in the release section for macos m1 I tried with cncjs-app-1.10.1-macos-arm64 and cncjs-app-1.10.0-macos-arm64 but both versions when trying to run the app says the app is damaged.

SPENCINAT0R, 2023-03-01:
1. Open Terminal on your macbook. 
2. Type this: "sudo spctl --master-disable"
3. Then, type this "xattr -cr " (ensure there is a space after "-cr") and drag and drop the CNCjs app into Terminal. 
4. Now try to open it and it should work. 
5. Ensure you type "sudo spctl --master-enable" to enable the security measures on your mac again. 

P.S. Please omit all quotes from the prompts in terminal.

closed 2024-10-17 by cheton

gadyet, 2024-11-07:
This no longer seems to work (tested on macOS 15.1 (24B83) "Sequoia"). "Run from anywhere" by itself does not fix the 'damaged' app message and xattr doesn't work.