cncjs/cncjs #818: Console widget doesn't print out configuration

VitaliiSadovshikov, 2023-02-27:
#### Description

I have stumbled across the following problem. If I have not interacted with CNCjs for a while, approximately one minute maybe, the Console widge will still receive my commands and executes them whereas it will not spit out backward any information about the machine. For instance, if I would type in "$X/" to the Console I would get nothing. The same issue with other widgets, for instance, "Axes" widget would not display the current position of the machine if I would leave the CNCjs at rest for a while.

#### Versions
- CNCjs: 1.10.1

#### How Do You Install CNCjs?
- [x] Download the CNCjs Desktop Application

#### CNC Controller
- [x] Grbl

#### Hardware
- [x ] Desktop or Laptop

#### Operating System
- [x] ] Mac

MitchBradley, 2023-02-27:
$X/ sounds like a FluidNC command.  What version of FluidNC?

VitaliiSadovshikov, 2023-03-01:
> $X/ sounds like a FluidNC command. What version of FluidNC?

Assume this is the type of command specific to [the xPro V5 controller](https://github.com/Spark-Concepts/xPro-V5/wiki/Changing-settings#grbl-commands) (not generic grbl).

MitchBradley, 2023-03-01:
Xpro v5 can run either Grbl_Esp32 or its successor FluidNC.  All of the Grbl_Esp32 developers have switched over to working on FluidNC. I am a lead developer on FluidNC.  If you connect to XPro V5 with a serial terminal and reset the controller, you can find out which version of the firmware is running.

VitaliiSadovshikov, 2023-03-01:
> Xpro v5 can run either Grbl_Esp32 or its successor FluidNC. All of the Grbl_Esp32 developers have switched over to working on FluidNC. I am a lead developer on FluidNC. If you connect to XPro V5 [...]

Is this what you've asked: 
"
> $Firmware/Info
FW version:1.3a (20220621) # FW target:grbl-embedded  # FW HW:Direct SD  # primary sd:/sd # secondary sd:none # authentication:no # webcommunication: Sync: 81:192.168.0.1 # hostname:CNC_xPRO_V5(AP mode) # axis:3
"

MitchBradley, 2023-03-01:
Consider upgrading to FluidNC.  Grbl_Esp32 has at least one [serious problem that can damage your hardware](https://github.com/bdring/FluidNC/issues/723#issuecomment-1445273667).

MitchBradley, 2023-03-01:
Another possible problem: `# hostname:CNC_xPRO_V5(AP mode)` .  ESP32 can be unstable when using AP mode for a long time.  The Espressif AP mode firmware appears to leak memory.  The Grbl_Esp32/FluidNC developers recommend using AP mode only for initial setup, then switching to STA mode or radio-off mode for production use.  If you need to use WiFi but do not have it in the machine area, one solution is to use a dedicated WiFi router that does not have to be connected to an internet backhaul.  Old/slow routers are fine for this purpose.  Many people have ones that have been removed from service that can be repurposed.

synman, 2023-03-22:
Unless someone breathes some life into CNCjs, you may want to start looking at other senders to meet your needs, especially when it comes to FluidNC.   

OpenBuilds was working on providing FluidNC support for their "CONTROL" sender the last time I spoke with them as their new BlackBox X32 runs FluidNC better than the bundled version of GrblHAL they provide.

My own little Octoprint plugin "Better Grill Support" also fully supports FluidNC.

And of course, the WebUI works just fine too and is getting a lot of support from the FluidNC team.

Hit me up on the side if you have any XPro specific questions as well.  It is a "been there, done that" for me if you require assistance.

MitchBradley, 2023-03-22:
> Maybe someone is breathing new life into it?

gSender is a maintained fork of CNCjs that, I think, has Cheton's blessing.

VitaliiSadovshikov, 2023-03-23:
> Consider upgrading to FluidNC

But how can I carry out it? Since, apparently Spark-Concepts has merely Grbl_Esp32 firmware options. Probably I need to wait FluidNC for them. I've inquired them about but still is pending.

MitchBradley, 2023-03-23:
You could have a separate controller for the extra rotary axis, but it would not be possible to have coordinated motion with that axis and others.  If you use the rotary to set an angle, then run a program, then change the angle, that might be okay.

Alternatively, you could do the dual-Y the old way by putting two motors on one driver.  Separate drivers is better, but ganging motors on one driver does work and a lot of machines do it that way.

VitaliiSadovshikov, 2023-03-23:
> Alternatively, you could do the dual-Y the old way by putting two motors on one driver. Separate drivers is better, but ganging motors on one driver does work and a lot of machines do it that way.

Thank you very much! Cool to know it! I thought about it but I was not sure that it's right connection. Because this way there maybe not enough power/corren on single slot. But you gave me confidence

> If you use the rotary to set an angle

Sorry, I am not sure what do you mean here