cncjs/cncjs #823: 3D view grid size

ramaxa9, 2023-03-24:
### Discussed in https://github.com/cncjs/cncjs/discussions/816

<div type='discussions-op-text'>

<sup>Originally posted by **ramaxa9** February 23, 2023</sup>
Hello!
The grid size is 600x600 mm. I set an a profile with my machine size 2000x1000 mm, but the grid size is still 600x600. Is it possible to increase grid size? </div>

cheton, 2023-03-26:
Currently, the grid size in the `Visualizer` component cannot be changed as it is hardcoded in the current releases. 

https://github.com/cncjs/cncjs/blob/a0ccc1685c52486b2fcde55b1e76363c97aac638/src/app/widgets/Visualizer/Visualizer.jsx#L35-L40

However, you can still define visible limits for the 3D visualizer and determine the size of your working space by configuring the machine profile. I can realize that the grid size should be automatically resized according to the machine profile, but this workaround can be used until new changes are made.

See the attached images for reference.

[image]

[image]

ramaxa9, 2023-03-27:
@cheton i have profile with my dimensions
this is not an a critical issue, it is quality of life fiture xD

and the camera cut portion of a model in 3d View

[image]