cncjs/cncjs #826: Docker Image throws: "Error: Cannot find module '@babel/polyfill'" on start

nick-filmriss, 2023-04-03:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [ ] Download the CNCjs Desktop Application
- [x] Docker

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [x] Linux (both tested)

cheton, 2023-04-08:
To resolve the issue, I suggest upgrading CNCjs to version [v1.10.1](https://github.com/cncjs/cncjs/releases/tag/v1.10.1) or a newer release. The latest version includes a fix for the "@babel/polyfill module not found" issue reported in version 1.10.0.

closed 2023-04-08 by cheton

nick-filmriss, 2023-04-10:
Thanks for the quick response.
I am using the latest docker image: https://hub.docker.com/r/cncjs/cncjs/ and still get the error:

`internal/modules/cjs/loader.js:905
  throw err;
  ^

Error: Cannot find module '@babel/polyfill'
Require stack:
- /opt/cncjs/entrypoint
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:902:15)
    at Function.Module._load (internal/modules/cjs/loader.js:746:27)
    at Module.require (internal/modules/cjs/loader.js:974:19)
    at require (internal/modules/cjs/helpers.js:101:18)
    at Object.<anonymous> (/opt/cncjs/entrypoint:3:1)
    at Module._compile (internal/modules/cjs/loader.js:1085:14)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
    at Module.load (internal/modules/cjs/loader.js:950:32)
    at Function.Module._load (internal/modules/cjs/loader.js:790:12)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:75:12) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ '/opt/cncjs/entrypoint' ]
}`

Does the Dockerimage use an old CNCJS version?

emcniece, 2023-05-17:
I think this is covered in https://github.com/cncjs/cncjs/issues/829 - got a PR coming in.

cheton, 2023-05-18:
New images are updated at https://hub.docker.com/repository/docker/cncjs/cncjs/general

[image]