cncjs/cncjs #827: Linux Electron app version v1.10.2 keeps returning: Error opening serial port '/dev/ttyS0'

SirGeekALot, 2023-04-07:
#### Description

I cannot connect to the serial port on Linux using the Electron app version 1.10.2 of CNC.js.  I tried my wife's Windows 11 laptop with the same version of the Electron app (1.10.2) and I verified I was able to connect and jog the machine successfully.  I set  both DTR and DTS on the Windows version to get it running, so I tried the same on the Linux box, but that did not help.

I keep getting the error: Error opening serial port '/dev/ttyS0' 

I get the same error with any one of the ports listed (ttyS0 through ttyS31).

I have power cycled the CNC router and restarted the CNC.js Electron app several times, but I'm not having any luck. I'm not sure how to troubleshoot this, but I am open to suggestions and will provide any debug info I can to help create a fix. Also, thanks for your work on this! I have been using it for several years, but my former Windows 7 Intel PC host had gotten too old so I wiped it and installed Linux Mint (which I think is great BTW), and that is why I am running it on Linux. I am a former software dev and reasonably good with Linux, but my knowledge of serial ports on Linux is...not great.

#### Versions

- CNCjs: 1.10.2
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application. I installed the electron app, version 1.10.2

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System
comfortable 
- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

MitchBradley, 2023-04-07:
Ask Google how to find serial ports on Linux. Once you find it make sure that you have permissions to open it. Google knows that too

MitchBradley, 2023-04-07:
Hint: ttySn often means a direct connect port, of the type that PCs no longer have. USB serial ports often have different names.  Different distros can have different naming conventions, which is just one of the reasons why Linux is such a nightmare to support

cheton, 2023-04-08:
It might probably a permission denied error. You can try adding the user account to the group owner of the `/dev/ttyUSB0` device.

The solution can be found on Stack Overflow:
https://stackoverflow.com/questions/68196282/arduino-ide-gives-cant-open-device-dev-ttys0-permission-denied-error

SirGeekALot, 2023-04-09:
> The solution can be found on Stack Overflow:

That link helped a lot! I had 2 problems: 1) I needed to switch the plug to a different USB port, and 2) I needed to reboot after adding my user account to the dialout group. I am now up and running again! Thanks!

closed 2023-04-09 by SirGeekALot