cncjs/cncjs #828: Features request: Widget or button to "Hold" gcode

Silver8411, 2023-04-10:
#### Description
Features request what if theirs widget or bigger button when gcode is running with red and "hold". So when things go soft "south" you can stop the gcode in hold mode via red button or spacebar ( spindle stops and lifts to safe height) and when its on hold that nice red button turns to 2 buttons. One as "resume" and other "stop". resume slowly lowers till top height in gcode and then continues with gode. Stop will aborts and goes home.

I do have e-stop for the very worse but this shut downs everything and needs to start all over again. But have had many times where i could save part or bit if it would just hold, i can do fixing the issue and then it resumes. 

Right now. those play, pause and stop button are so small there. when actually you need to react. its rely hard to click them. 

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2023-04-11:
The cncjs-shopfloor-tablet UI has almost exactly what you are asking for

Silver8411, 2023-04-11:
Is it possible to use or add that function to pc cncjs version? i have dedicated laptop for controlling cnc and it runs pc version of cncjs. Dont have there any other devises to use it on machine

MitchBradley, 2023-04-11:
Yes you can run cncjs-shopfloor-tablet from the PC app.  Do this:

1. Go to https://github.com/cncjs/cncjs-shopfloor-tablet/releases/tag/v2.0.0 and download the release asset file [cncjs-shopfloor-tablet-2.0.0.zip](https://github.com/cncjs/cncjs-shopfloor-tablet/releases/download/v2.0.0/cncjs-shopfloor-tablet-2.0.0.zip)
2. Extract that .zip file to some folder on your PC.  On my PC, Windows Explorer extracts by default to a subfolder of Downloads, but you can choose any folder.
3. Use a text editor like notepad to edit the file \Users\YOURUSERNAME\.cncrc .  Add a "mountPoints" section, or if one already exists, add an additional entry to it.  The file probably starts with something like this:
```
{
    "state": {
        "checkForUpdates": true,
```
You need to add a new section so it looks like this:
```
{
    "mountPoints": [
        {
            "route": "/tablet",
            "target": "C:/Users/YOURNAME/Downloads/cncjs-shopfloor-tablet-2.0.0/cncjs-shopfloor-tablet-2.0.0"
        }
    ],
    "state": {
        "checkForUpdates": true,
```
Replace YOURNAME with your own user name - or replace the entire target: path with the location where you extracted the .zip file.  Save the modified file.

Then restart the cncjs app and look in the View menu:
[image]
If you click on the /tablet item down at the bottom, it should take you to the tablet UI in another browser.

It it doesn't work, you probably did not follow the instructions precisely.  The syntax of the .cncrc file is very picky so you must get all the braces, brackets, comma, and quotes just as shown.