cncjs/cncjs #829: Latest Docker image throws: Error: Cannot find module '@babel/polyfill'

nick-filmriss, 2023-04-15:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.10.2 (Docker latest)
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [ ] Download the CNCjs Desktop Application
- [x] Docker

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [x] Linux

I am using the cncjs:latest docker container. 
(Also present in older images)
The container throws following error on start:

`internal/modules/cjs/loader.js:905
throw err;
^

Error: Cannot find module '@babel/polyfill'
Require stack:

    /opt/cncjs/entrypoint
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:902:15)
    at Function.Module._load (internal/modules/cjs/loader.js:746:27)
    at Module.require (internal/modules/cjs/loader.js:974:19)
    at require (internal/modules/cjs/helpers.js:101:18)
    at Object. (/opt/cncjs/entrypoint:3:1)
    at Module._compile (internal/modules/cjs/loader.js:1085:14)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
    at Module.load (internal/modules/cjs/loader.js:950:32)
    at Function.Module._load (internal/modules/cjs/loader.js:790:12)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:75:12) {
    code: 'MODULE_NOT_FOUND',
    requireStack: [ '/opt/cncjs/entrypoint' ]
    }`

Does the Dockerimage use an old CNCJS version?

evdel, 2023-04-27:
I see the same incident. Any help would be much appreciated.

evdel, 2023-04-27:
Where can we read the Dockerfile ?

evdel, 2023-04-27:
Apparently this is not new and was fixed here https://github.com/cncjs/cncjs/releases/tag/v1.10.2

randomiq, 2023-05-12:
I can confirm that the latest image for me has this error, running v1.10.2:
```
root@buffalo:/srv/cncjs# docker ps | grep cncjs
4836893bc876   cncjs/cncjs:v1.10.2             "/opt/cncjs/entrypoi…"   3 hours ago   Restarting (1) 12 seconds ago                                                                                                                     cncjs-cnc-1
```

Docker Log:
```
Error: Cannot find module '@babel/polyfill'
Require stack:
- /opt/cncjs/entrypoint
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:902:15)
    at Function.Module._load (internal/modules/cjs/loader.js:746:27)
    at Module.require (internal/modules/cjs/loader.js:974:19)
    at require (internal/modules/cjs/helpers.js:101:18)
    at Object.<anonymous> (/opt/cncjs/entrypoint:3:1)
    at Module._compile (internal/modules/cjs/loader.js:1085:14)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
    at Module.load (internal/modules/cjs/loader.js:950:32)
    at Function.Module._load (internal/modules/cjs/loader.js:790:12)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:75:12) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ '/opt/cncjs/entrypoint' ]
}
internal/modules/cjs/loader.js:905
  throw err;
  ^
```
The above repeats over and over as the container restarts.

randomiq, 2023-05-13:
Did a bit more mucking about in the container and looking at the past commits.  The Docker entrypoint file needs to have the same changes as the cncjs in https://github.com/cncjs/cncjs/pull/803. Specifically removing .default from the end of:
`const launchServer = require('/opt/cncjs/server-cli')
`

Also the @babel/polyfill module needs to be installed.  Ideally this is done at build time of the Dockerfile.  Otherwise npm install --save @babel/polyfill inside the container can do it temporarily for testing.

With those 2 modifications the entrypoint scripts starts cncjs up fine.

My node/npm/yarn experience is rough at the moment... maybe tomorrow night will dust it off and see about a PR.

randomiq, 2023-05-16:
Just to give an update on the Dockerfile and building everything from scratch inside the build container.  I am able to build from source a docker container by making a handful of changes to the Dockerfile.  However, I think I'm doing too many steps.  Comments/background appreciated on the RUN lines below:

```
#######################################
# Mucking about and figuring out how this
# could be built fully in a Docker container
# .....
# Copy our source into the container - This means that cncjs-source is a copy of the git repo
COPY . $BUILD_DIR/cncjs-source

# Install Yarn, then cncjs modules, then build
WORKDIR $BUILD_DIR/cncjs-source
RUN npm install -g yarn
RUN yarn install
RUN npm run build-prod

# Copy the built dist/cncjs and entrypoint over to the target cncjs dir
RUN cp -a dist/cncjs $BUILD_DIR/ && cp entrypoint $BUILD_DIR/cncjs

# Don't understand enough, so I think there redundant commands above and below...
# npm install this, yarn install that...
WORKDIR $BUILD_DIR/cncjs
RUN npm install
RUN npm install --save @babel/polyfill
RUN yarn --production
```

emcniece, 2023-05-16:
Hey @randomiq! Thanks for the ongoing efforts, you're doing great 😁 those Dockerfile comments help explain your approach nicely. There are a few redundancies in your Dockerfile, I won't have time to really dig in to this until later but maybe I can point a few things out right now.

Keeping image layers to a minimum for a final product is often desired. Perhaps as you solidify your changes you can move toward aggregating the `RUN` statements`, or implement a multi-stage "builder pattern". There are examples of this in the current Dockerfile: [aggregate operations](https://github.com/cncjs/cncjs/blob/master/Dockerfile#L33), [builder pattern](https://github.com/cncjs/cncjs/blob/master/Dockerfile#LL36C21-L36C21).

This repo focuses on using `yarn` over `npm`, which is only used to install `yarn`. The `yarn` command does (roughly) the same thing as `yarn install`, `yarn --production`, and `npm install`. You should only have to build the project once. I'm not totally sure if `npm run build-prod` is needed - I don't see that in the existing Dockerfile, and I have a suspicion that `yarn --production` might do more than just download packages.

Your first stage loads the project files into `$BUILD_DIR/cncjs-source`, which is then installed and built - all good. The later stage changes `WORKDIR $BUILD_DIR/cncjs` though, what files are present in this directory at this time? I don't think `npm install` will do anything without a `package.json` file, but maybe that's included in `dist/cncjs`. Regardless, the project is already built at this stage and the later `npm install` and `yarn --production` steps should be redundant.

**Here's the real nugget of gold**: `npm install --save @babel/polyfill`.

This ticket was opened to address this missing package, and from what I can tell _it should't be missing_. `yarn --production` will install packages, but only the main dependencies and not `devDependencies` from package.json. Sometimes polyfills are only included in devDependencies because they're used for transpilation and not needed for post-build runtime. The confusing bit is that this package is included in _both_ sections:

- https://github.com/cncjs/cncjs/blob/master/package.json#L129
- https://github.com/cncjs/cncjs/blob/master/package.json#L278

By explicitly installing this package you're guaranteeing that it is present in the image. We should be able to get the same results (ie. a working Docker image) by adding this install step to the existing Dockerfile somewhere in the [final stage](https://github.com/cncjs/cncjs/blob/master/Dockerfile#L36). Executing `npm install` with `--save` will update package.json, which shouldn't be necessary as package.json already has it.

To troubleshoot this issue I might suggest splitting the existing Docker image in half and inspect `node_modules` and the `dist` directories at the end of each stage. `@babel/polyfill` should be present somewhere, either in a directory or in a transpiled bundle. If it _is_ present and the image still throws this error, I'd question the runtime (nvm) as perhaps the environment variables that set the Node resources (https://github.com/cncjs/cncjs/blob/master/Dockerfile#L41-L42C56) don't have the correct packages or aren't being used as expected.

I can't guarantee a timeline for helping out (things are hectic at the moment) but I'm following along and I'll do what I can to help answer questions!

randomiq, 2023-05-16:
Thanks for the feedback.  I should have prefixed that my above changes were to the existing Dockerfile. So the multiple RUN lines are in the 'build' container where I'm ok with having many layers as we don't actually keep'em once we get what we want (dist/cncjs).

Rereading my Dockerfile and reading in your comments above here's the whole Dockerfile with my understanding on what/why I have some of the lines in there:

```
# BUILD STAGE
FROM debian:bullseye as build-stage

ENV BUILD_DIR /tmp/build
ENV NVM_DIR /root/.nvm
ENV NODE_VERSION v14.21.2
ENV NODE_ENV production
ENV NODE_PATH $NVM_DIR/$NODE_VERSION/lib/node_modules
ENV PATH $NVM_DIR/versions/node/$NODE_VERSION/bin:$PATH

RUN apt-get update -y && apt-get install -y -q --no-install-recommends \
  apt-utils \
  build-essential \
  ca-certificates \
  python3 \
  python3-pip \
  curl \
  git \
  udev

RUN git clone https://github.com/nvm-sh/nvm.git "$NVM_DIR" \
  && cd "$NVM_DIR" \
  && git checkout `git describe --abbrev=0 --tags --match "v[0-9]*" $(git rev-list --tags --max-count=1)` \
  && . "$NVM_DIR/nvm.sh" \
  && nvm install "$NODE_VERSION" \
  && nvm alias default "$NODE_VERSION" \
  && nvm use --delete-prefix default

#######################################
# Mucking about and figuring out how this
# could be built fully in a Docker container
# .....
# Copy our source into the container
COPY . $BUILD_DIR/cncjs-source

# Install Yarn, then cncjs modules, then build
WORKDIR $BUILD_DIR/cncjs-source
RUN npm install -g yarn  # Install yarn as it's used in build-prod
RUN yarn install         # Install modules in yarn.lock ? or does yarn install modules in package.json
RUN npm run build-prod   # Make the dists dir and run all our build commands needed

# Copy our dist/cncjs output from above and entrypoint over to a target directory
RUN cp -a dist/cncjs $BUILD_DIR/ && cp entrypoint $BUILD_DIR/cncjs

# Don't understand enough, so I think there redundant commands above and below...
# npm install this, yarn install that...
WORKDIR $BUILD_DIR/cncjs
RUN npm install           # The dists/cncjs (aka cncjs) dir has a package.json whys is this needed here?
RUN npm install --save @babel/polyfill # Adding in babel/polyfill here, but why can't this already done above in cncjs-source?
RUN yarn --production     # No clue what this does, throwing kitchen sink.
#
#######################################

# FINAL STAGE
FROM debian:bullseye

ENV NVM_DIR /root/.nvm
ENV NODE_VERSION v14.21.2
ENV NODE_ENV production
ENV NODE_PATH $NVM_DIR/$NODE_VERSION/lib/node_modules
ENV PATH $NVM_DIR/versions/node/$NODE_VERSION/bin:$PATH

RUN apt-get update -y && apt-get install -y -q --no-install-recommends \
  apt-utils \
  ca-certificates \
  udev

WORKDIR /opt/cncjs
EXPOSE 8000
ENTRYPOINT ["/opt/cncjs/entrypoint"]

COPY --from=build-stage /root/.nvm $NVM_DIR
COPY --from=build-stage /tmp/build/cncjs /opt/cncjs
```

emcniece, 2023-05-16:
Aha, ok that context helps. I see now that the existing Dockerfile starts out by copying `./dist/cncjs` into the container, so the build process would have happened separately, outside the `docker build` context. Got it. Your `npm run build-prod` step makes sense now, the Yarn equivalent I think would be `yarn run build-prod`. 

> Install modules in yarn.lock ? or does yarn install modules in package.json

Yes and yes 👍 it'll use yarn.lock first and fall back on package.json for upgrades. If Yarn works like NPM does, `yarn --production` would _only_ use yarn.lock.

> # Copy our dist/cncjs output ...

At this stage, the project is built and the assets are being copied to the build directory. I bet the image will work if the next few lines are simplified to:

```
# Don't understand enough, so I think there redundant commands above and below...
# npm install this, yarn install that...
WORKDIR $BUILD_DIR/cncjs
# RUN npm install           # The dists/cncjs (aka cncjs) dir has a package.json whys is this needed here?
# RUN npm install --save @babel/polyfill # Adding in babel/polyfill here, but why can't this already done above in cncjs-source?
# RUN yarn --production     # No clue what this does, throwing kitchen sink.
```

> Adding in babel/polyfill here, but why can't this already done above in cncjs-source?

That's the real question. The polyfill package should already be installed! Mysterious.

randomiq, 2023-05-16:
I see differing references to babel/polyfill:

```
root@buffalo:/data/Projects/cncjs# grep 'babel/polyfill' *.json *.lock
package.json:    "test": "tap test/*.js --no-timeout --node-arg=--require --node-arg=@babel/register --node-arg=--require --node-arg=@babel/polyfill",
package.json:    "coveralls": "tap test/*.js --coverage --coverage-report=text-lcov --nyc-arg=--require --nyc-arg=@babel/register --nyc-arg=--require --nyc-arg=@babel/polyfill",
package.json:    "@babel/polyfill": "~7.4.3",
package.json:    "@babel/polyfill": "~7.4.3",
package-lock.json:        "@babel/polyfill": "~7.4.3",
package-lock.json:        "@babel/polyfill": "~7.4.3",
package-lock.json:        "@babel/polyfill": "^7.0.0",
package-lock.json:    "node_modules/@babel/polyfill": {
package-lock.json:      "resolved": "https://registry.npmjs.org/@babel/polyfill/-/polyfill-7.4.4.tgz",
package-lock.json:      "deprecated": "🚨 This package has been deprecated in favor of separate inclusion of a polyfill and regenerator-runtime (when needed). See the @babel/polyfill docs (https://babeljs.io/docs/en/babel-polyfill) for more information.",
package-lock.json:    "node_modules/@babel/polyfill/node_modules/core-js": {
yarn.lock:    "@babel/polyfill": ^7.0.0
yarn.lock:"@babel/polyfill@npm:^7.0.0":
yarn.lock:  resolution: "@babel/polyfill@npm:7.12.1"
yarn.lock:"@babel/polyfill@npm:~7.4.3":
yarn.lock:  resolution: "@babel/polyfill@npm:7.4.4"
yarn.lock:    "@babel/polyfill": ~7.4.3
```

version ~7.4.3 or ^7.0.0 or 7.4.4 or 7.12.1....

_Insert table flip..._

emcniece, 2023-05-17:
These references are dependencies stating a requirement for a version range of the polyfill library. Npm will handle dependency resolution based on the semantic version range of each of these requirements. Running `npm ls` or `npm ls @babel/polyfill` can help visualize what and where is using a particular version.

Semver range cheat sheet: https://devhints.io/semver

Hearing that 7.12.1 is in cncjs/node_modules/@babel/polyfill is interesting - package.json should be driving this particular version, and it specifies `~7.4.3` which should only accept versions up to `7.5.0`, a minor version bump. The explicit command `RUN npm install --save @babel/polyfill` is installing the latest version, [7.12.1](https://www.npmjs.com/package/@babel/polyfill).

This might be part of the problem, and you might have found the solution. Maybe 7.4.3 no longer works with some other library in this project, and by upgrading to 7.12.1 it makes the image work again. To test this hypothesis we could build the Dockerfile with `RUN npm install --save @babel/polyfill@7.4.3` and see if it fails.

emcniece, 2023-05-17:
Doing some investigation right now. Interesting find: the Docker container starts up if `require('@babel/polyfill');` is removed from `./entrypoint`.

Searching the repo for the history of the polyfill package we find this PR linked to a similar error report: https://github.com/cncjs/cncjs/pull/803/files. Several libraries were updated, and `require('@babel/polyfill');` was removed from many places. I suspect that it should have been removed from `./entrypoint` and it was missed.

After modifying entrypoint and building the existing Dockerfile I'm able to get the container running, but accessing the browser UI displays an error and terminal reports:

```
Error: Failed to lookup view "index.hbs" in views directories "/opt/cncjs/app" or "/opt/cncjs/server/views"
```

Progress, either way.

emcniece, 2023-05-17:
PR opened: https://github.com/cncjs/cncjs/pull/831

It doesn't fully work yet, there's a problem with the handlebars template rendering I think. Getting close though.

closed 2023-05-18 by emcniece

cheton, 2023-05-18:
New images are updated at https://hub.docker.com/repository/docker/cncjs/cncjs/general

[image]

emcniece, 2023-05-18:
And it works!

```
docker pull cncjs/cncjs:v1.10.2
docker run -d -p 8000:8000 cncjs/cncjs:v1.10.2
```

... and http://localhost:8000 renders the UI 😁 

I'm still confused about the index.hbs file and manual builds, but happy to report that the Docker image should be functional again.

randomiq, 2023-05-18:
Yep, can confirm that the issue is fixed.  Thank you again.

randomiq, 2023-05-18:
Also now that I see that circleci is used I peeked at the https://github.com/cncjs/cncjs/blob/master/.circleci/config.yml file and it lays out all the commands run outside of the docker container build to get it to built.  Not convenient for those who don't want to install node/yarn/devEnv on their machine to build the docker image.  Now I've got some CNC projects to work on!

cheton, 2023-05-18:
> Also now that I see that circleci is used I peeked at the https://github.com/cncjs/cncjs/blob/master/.circleci/config.yml file and it lays out all the commands run outside of the docker container [...]

You can see my comment at https://github.com/cncjs/cncjs/pull/831#discussion_r1197329829

Yes, it's a bit confusing. To simplify this, I will check if it is possible to streamline the build process by running the `yarn build` command within the Dockerfile.