cncjs/cncjs #830: Feature Request for Events

crackinx, 2023-05-10:
any way you can add an event for alarm? I would like to be able to run a command through IFTTT to send me a text on alarm. Also my vacuum is run from a smart plug to start and stop with the job, I would also like it to turn off on alarm since the system stop running anyway. both of which I can do with an event on alarm with an IFTTT webhook. Thanks!!

Fusseldieb, 2023-06-30:
You could do that with a custom plugin.

There's a demo plugin which you could modify. It basically runs separately and connects to the CNCjs instance over some port. It then can listen for any change/update that's happening.

You could modify it to call IFTTT when it detects a certain event or gcode.

crackinx, 2023-06-30:
Not 100% sure how to go about that, but I'll look into it and give it a shot.