cncjs/cncjs #833: Bump Node 7.x to 14.x in rpi-setup-guide

Fusseldieb, 2023-07-05:
#### Description

I recommend bumping the Node Version to **14.x** on the below guide, as older ones are unsupported and don't run properly anymore.

https://cnc.js.org/docs/rpi-setup-guide/ (It recommends 7.x)

Line: 
`curl -sL https://deb.nodesource.com/setup_7.x | sudo -E bash -  # Install NodeJS v7`

To:
`curl -sL https://deb.nodesource.com/setup_14.x | sudo -E bash -  # Install NodeJS v14`

I usually get a few deprecation warnings, but it works nicely with an up to date system.

#### Versions

- CNCjs: 1.10.3
- Node.js: 14.x
- NPM: 9.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Smoothieware
- [ ] TinyG/g2core
- [x] Marlin

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

tarheele, 2023-09-26:
Agree.  Tried 14 on the May 3rd 2023 RPIOS with desktop build and got "Error: Cannot find module 'timers/promises'"

moved to 16 and viola!

===========

https://www.raspberrypi.com/software/operating-systems/
Raspberry Pi OS with desktop
Release date: May 3rd 2023
System: 64-bit
Kernel version: 6.1
Debian version: 11 (bullseye)

Mawkes, 2024-07-08:
After running "curl -sL https://deb.nodesource.com/setup_16.x | sudo -E bash -  # Install NodeJS v16"

...this command:
# Update Node Package Manager (NPM)
sudo npm install npm@latest -g

...results in:
user@orangepizero:~$ sudo npm install npm@latest -g
npm ERR! code EBADENGINE
npm ERR! engine Unsupported engine
npm ERR! engine Not compatible with your version of node/npm: npm@10.8.1
npm ERR! notsup Not compatible with your version of node/npm: npm@10.8.1
npm ERR! notsup Required: {"node":"^18.17.0 || >=20.5.0"}
npm ERR! notsup Actual:   {"npm":"8.19.4","node":"v16.20.2"}

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2024-07-08T09_55_03_600Z-debug-0.log

AustinSaintAubin, 2024-07-11:
Updated the [cncjs-pi-raspbian](https://github.com/cncjs/cncjs-pi-raspbian) install script. 
See README.md for guide.

Works with:
- NodeJS: v18.19.0
-  NPM: v9.2.0

closed 2024-10-07 by cheton