cncjs/cncjs #836: How to make settings persistant?

Elespecia, 2023-07-29:
#### Description

I'm sick of having to make all the settings every now and then, and everytime I redeploy the Docker-image (yes, I'm running the contianerversion).

I fail to see any file that I can bind with Docker but also have limited knowledge of you NPM and NodeJS work.

#### Versions

- CNCjs: All versions (Currently at 1.10.3)

#### How Do You Install CNCjs?

- [ ] NPM
- [ ] Download the CNCjs Desktop Application
- [X] Docker-compose

```
version: "3.7"
services:
  cncjs:
    container_name: CNCjs
    image: cncjs/cncjs:latest
    ports:
      - "8005:8000"
    volumes:
      - /docker/CNCjs:/cncjs
    network_mode: bridge
    hostname: cncjs
    restart: unless-stopped
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0

```
#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [X] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [X] Linux

thijstriemstra, 2023-08-28:
Good question; but I think it's a session/cookie thing, meaning it's only persisted in the browser. I'd love to be proven wrong though.

Elespecia, 2023-08-28:
I think you're right.
That's not ideal with machine settings, macros and configurations.

onezero1010101, 2024-04-17:
So I think I might have this fixed, try these volumes and see if that helps.  The .cncjs-sessions is a folder, .cncrc is a file.  You will need to create a blank file for the .cncrc, or it will create a folder

```
name: cncjs
services:
  cncjs:
    container_name: cnc
    image: cncjs/cncjs:latest
    privileged: true
    ports:
      - 80:8000
    volumes:
      - ./cncjs/.cncjs-sessions:/root/.cncjs-sessions
      - ./cncjs/.cncrc:/root/.cncrc

```

Elespecia, 2024-04-18:
Thanks for the advice but there is no files in my .cncjs-sessions and the .cncrc only involve a token to a session where the settings are stored.
To not loose all my settings once again I need to extract the settings prior to recreating the container but no settings are to be found.

However, locating the secret is one of two keys for persistance I think.

Elespecia, 2024-04-18:
Perhaps the key is to figure out where cncjs-app-1.10.3.json is located and after exported have it bound.
But where is is located? I think the devs can tell us in a second if they look at this issue.

stb1Cluny, 2025-01-09:
I'm having some success adding settings to, and mounting, a .cncrc file as was suggested by @onezero1010101 and shown in some other issues and examples.
```
volumes:
  - ./cncjs/.cncrc:/root/.cncrc
```

It seems like you can do quite a bit with it. There's an example in the source here: https://github.com/cncjs/cncjs/blob/1fa345c2d4f746a2bb6849049eff6147a93459ca/examples/.cncrc