cncjs/cncjs #839: shuttlexpress pro 2 cncjs connection issue

slausecrx, 2023-08-15:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.10.3x
- Node.js: 6.x  (not sure)
- NPM: 5.x (not sure)

#### How Do You Install CNCjs?

- [] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller xpro v5 

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [ x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ x] Windows
- [ ] Mac
- [ ] Linux

i hope i am doing this right. very first post on github.   i have been searching for weeks and i cant get anywhere on my own.

my prob is i have a cnc router that i would like to use my shuttlexpress on. i have set all the buttons on the device but i can not get it to communicate with cncjs i am hoping i can find some help here thanks!

slimline33, 2023-08-16:
Hast du die Software für den shuttle installiert und kannst du auch andere Programme damit steuern? Ich nutze den Pro und bei mir funktioniert es eigentlich ganz gut. Ich nutze auch i-macros für Chrome und dann noch ein Plugin um mir die Makros auf eine Taste legen zu können.

slausecrx, 2023-08-16:
i do have the shuttle software installed and on the software you can see the button depress and it shows my keyboard shortcuts that i gor of the cncjs github. i have not tried it on other programs. i will try that now. is there anything i need to do inside of cncjs to make this work or should it be plug and play? thanks so much for the response!!!!

slausecrx, 2023-08-16:
i have tried the shuttle with google chrome and it works as should. i have retried with cncjs and i get zero response from cncjs any help would be awesome thanKS!!!

slimline33, 2023-08-16:
Ok. Ich nutze cncjs unter Linux und rufe das unter Windows mit meinem Browser auf. Hast du ein custom Profil erstellt um cncjs.exe im Programm Verzeichnis mit deinem shuttle express zu nutzen? Du musst ein custom Profil verwenden und unten in der task Symbol leiste wird automatisch das Programm im Vordergrund gewählt. Ich denke es liegt an der automatischen Auswahl von shutter control

slausecrx, 2023-08-16:
i am using windows 10 pc connected to my cnc controller via usb type C. i have made a custom profile named it cncjs. i have set the values using the guthub for cncjs. in the shuttle software under the cncjs profile there is a "options button" inside that there is a "target application" i select cncjs file there i willl attach pics thanks again man!!

[image]
[image]
[image]

slimline33, 2023-08-19:
Entschuldigung da kann ich dir leider nicht helfen. Ich nutze cncjs im Browser. Da funktioniert es wunderbar!