cncjs/cncjs #840: Random halts every few seconds

jwaes, 2023-08-31:
#### Description

Excecution of my gcode gets interrupted regularly. It does not output an error but one or more smbols in the console.

```
Grbl 1.1h ['$' for help]
client> $$
[MSG:'$H'|'$X' to unlock]
$0=10 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=1 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=1 (Homing direction invert, mask)
$24=25.000 (Homing locate feed rate, mm/min)
$25=8000.000 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=2.000 (Homing switch pull-off distance, millimeters)
$30=255 (Maximum spindle speed, RPM)
$31=0 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=80.000 (X-axis travel resolution, step/mm)
$101=80.000 (Y-axis travel resolution, step/mm)
$102=250.000 (Z-axis travel resolution, step/mm)
$110=4000.000 (X-axis maximum rate, mm/min)
$111=4000.000 (Y-axis maximum rate, mm/min)
$112=500.000 (Z-axis maximum rate, mm/min)
$120=400.000 (X-axis acceleration, mm/sec^2)
$121=400.000 (Y-axis acceleration, mm/sec^2)
$122=10.000 (Z-axis acceleration, mm/sec^2)
$130=876.000 (X-axis maximum travel, millimeters)
$131=1000.000 (Y-axis maximum travel, millimeters)
$132=50.000 (Z-axis maximum travel, millimeters)
ok
> $120=200
ok
> $121=200
ok
client> $H
>G53X-850Y-950:ok
>G92X0Y0:ok
ok
kC᫊M�o
[GC:G1 54 G17 G21 G90 G94 M3 M9 T0 F8000 S5]
k
<Run|MPos:-818.963,-890.838,0.000|FS:21085|Pn:S|Ov:100,100,100|S>4
o
o
client> !
client> ~
k
client> !
client> ~
�
oS5
client> !
client> ~
Idle|MPo
```
As you can see above, it stops and stays in the playing state.
When i hit pause and then play it continues for a bit again.

Sometimes, it does halt with an error

```
client> !
client> ~
> G1 X223.738 Y70.713 (line=495)
error:1 (Expected command letter)
k
Idle|MPos:-821.900,-875.800,0.000|FS:0,5|Pn:S>
>
```

Then i do not have to hit pauze and play, but just play to continue.

The characters are sometimes high ascii chars, but most often just 'o' and 'k' ... so it might be related to this issue https://github.com/cncjs/cncjs/issues/822

#### Versions

- CNCjs: 1.10.3
- Node.js: 18.17.1
- NPM: 9.8.1

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

Setup:
cncjs running on rpi 4, connected with short shielded usb cable to Arduino Uno with a CNC shield v3.0

karoria, 2023-09-01:
Probably EMI causes this random inputs. What are all means of start and pause in your setup? Are they optoisolated if hard wired input buttons? Using any matrix keypad style inputs? Then they are highly prone to EMI. Use debouncing in code if so.

jwaes, 2023-09-02:
You can close the issue.

I replaced the Arduino Uno clone by a real one and the issue was fixed ! It is the CH340G chip that causes the issues.

https://github.com/grbl/grbl/wiki/Known-Bugs#usb-to-serial-transmission-errors

closed 2024-10-07 by cheton