cncjs/cncjs #842: [FR] Widget: Multipoint Leveling with Proceadural Extrapolation of Static values imported from GCODE file

rflulling, 2023-10-02:
#### Description

Feature Request to add a Widget for use in CNC. Where in the slicer is able to read the gcode for all top most values. Then adjust based actual measured data from supplied stock assumed to be non uniform. Matters only for machining where a TOP down method is selected. Where distance from TOP edge of Stock cannot be machined flush and machining must take into account deviation in the upper edge prior to start. 

Additional option request. Force behavior into Firmware. Some firmware have the capacity to handle this already. Some it must be done by the external controller. ie Marlin has many kinds of leveling built in, but GRBL depends on the controller. 
Advanced functionality: maintain the offset and needed tweeks to values in RAM, carry the adjustments all the way to end of the cut, behaving as if the bottom most edge deviates the same. I do not know of any firmware that can carry the leveling offsets through an entire cut since it requires a lot of RAM in a controller that has almost none compared to a PC. 

I suspect the Widget will need to run in its own thread. Pull the data from CNCjs, then hand it back for printing unless it should be sent to the serial terminal widget directly. 

#### Versions

- CNCjs: 1.10.2 DESKTOP
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [x] MARLIN 2.1.2.1
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device
- [X] BTT SKR Mini E3 V3.0.1

#### Operating System

- [ ] Not Applicable
- [x] 64bit Windows 10 Pro N 
- [ ] Mac
- [ ] Linux

MitchBradley, 2023-10-02:
You might want to look at the commit history to see how many features have been added in the last few years.

rflulling, 2023-10-19:
> You might want to look at the commit history to see how many features have been added in the last few years.

It is not so much that I expect anything. I am putting this out there. I found since posting some one did pots back in 2017 a way to do something very similar. with CNCjs, but not as a widget. They were using macros. 

My hope is simply to put it out there and maybe it will gain recognition. 

If I was skilled enough as a programmer ( and I am 100% not) then I would just write this myself.