cncjs/cncjs #845: zero out work offsets: Marlin

rflulling, 2023-10-21:
**#### Description**
When this function is invoked it generates a command `G10 L20 P1 Z0.` 
In marlin this command directs a firmware controlled filament Retraction. 
If a machine is set up with FW retraction not used since this is a CNC, then a user will see a "Unknown Command" in serial. 

What we really need is a `G92 Z0`, or `G92 X0 Y0`
Use of this command will set a virtual zero from wherever the head position is. 
[https://marlinfw.org/docs/gcode/G092.html](https://marlinfw.org/docs/gcode/G092.html)
There is also M428, but I am not familiar with it's use.
[https://marlinfw.org/docs/gcode/M428.html](https://marlinfw.org/docs/gcode/M428.html )

Sadly this is one of a few very specific differences between marlin and old school CNC specific gcode. 
Thankfully, it is not a show stopper and the needed commands can be built into a file or entered manually. 

**#### Versions**

- CNCjs: 1.10.2
- Node.js: 14.8.3
- NPM: 5.x

**#### How Do You Install CNCjs?**

- [ ] NPM
- [x] Download the CNCjs Desktop Application

**#### CNC Controller**

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
- [x] Marlin 2.x

**#### Hardware**

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

**#### Operating System**

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

wilcomir, 2023-11-04:
I too have the same issue. I have fixed this with a couple of macros, but it would really be great to build this into cncjs.

rflulling, 2026-01-21:
Bump. How would I go about trouble shooting this myself? Do I need to copy the entire source code or is this just one widget?

To be fair I think on a default marlin build updating work place offsets should not do anything. So making this functional does require the user to have compiled a copy of their firmware with offsets enabled in the configuration files. 

However there is no value added to executing a filament retraction when trying to set stock offsets and start positions, even on a stock version of Marlin.