cncjs/cncjs #851: Connection to Com port

JamesjoGibson, 2024-01-08:
Can not connect to Com 3 port in CNCjs  version 1.10.3

In the drop down Port menu there is never an option to connect -  see below greyed out

[image]

I Can connect via the CNC_xPRO_V5(AP mode) and make the CNC work on all 3 x axis - see below

[image]

I am connected from the CNC xPRO via USB cable
 
Versions

- CNCjs: 1.10.3
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl - CNC X pro V5

#### Hardware
Running on new Laptop

#### Operating System
[image]

What am I doing wrong

James

ipsum-coder, 2024-01-14:
Hi James, 
It seems you are using a port that is not being detected by the CNCjs Software. I highly recommend you to try the following steps: 
i) Close all the other softwares which are using the ports or have access to ports
ii) Replug the cnc
iii) Check if the usb cable is properly shielded
iv) Restart The Program

I hope it would work fine. If not try using the client version of cncjs. 

Regards,
Ritik AKA Ipsum Coder

MitchBradley, 2024-01-14:
Have you tried without rts/cts flow control?

MitchBradley, 2024-01-14:
Also, do you have the CP210x USB-Serial driver installed on your laptop?

JamesjoGibson, 2024-01-15:
Ta Titik
I have tried closing other ports – No luck there
Unplugged and reconnected multiple times
Restarted Both CNC and computer multiple times.
I have had suggestion that the CPX210 driver may be the issue, so trying that now

Thanks
James Gibson
Director
0427 290 323
aquawest.com.au<http://www.aquawest.com.au/>
***@***.***

closed 2024-10-07 by cheton