cncjs/cncjs #854: G-code checksums?

d01010101, 2024-02-05:
I would like to swap the boards in my cnc machine. I use ramps/grbl but there are occasional uncorrected transmission errors. I guess that cncjs does not support Marlin [checksums](https://reprap.org/wiki/G-code#.2A:_Checksum)?

SENT: N24 G1 X120.405 Y76.035 E2.23535*97
RECV: Error:checksum mismatch, Last Line: 23
RECV: Resend: 24
RECV: ok

naveedyassin, 2025-04-10:
Since you’re already familiar with GRBL/RAMPS, switching to something more robust like:
	•	CNC Shield + Arduino Uno + GRBL
	•	GRBL on STM32 or ESP32 (like FluidNC)
	•	GRBL_ESP32 — highly reliable and network-capable
	•	GrblHAL (Teensy, STM32, etc.) — advanced and checksum-safe

These setups work seamlessly with CNCjs and don’t require checksums.

MitchBradley, 2025-04-10:
Note that GRBL_ESP32 is no longer supported; FluidNC is its successor.