cncjs/cncjs #863: CNCJS needs view reload on start (Desktop App on Win10)

Chris-the-electrician, 2024-05-13:
#### Description

CNCJS (64bit version) shows a blank (white) screen after start, this happens in about 99%.
After "View -> reload" or "View -> force reload" everything is OK. This behavior at application startup is a little bit annoying.
Behavior appears on all of my devices (HP Elitebook, ASUSpro, ... - all devices are running Win10 pro 64bit).
Does anyone know a fix for that?
This is the only problem I found. In all other matters, CNCJS is working great. It's the best GRBL interface I found so far, very clear and functional.
Just in 3D view, when a .NC file has a big number of very short machine movements, the 3D view freezes (but very rare), so I just disable the 3d view before starting a job, I always simulate a job in another sim program before milling. But the 3D view is still good to check the path limits before starting a job.

#### Versions

- CNCjs: 1.10.x
- Node.js: 20.11.0
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl (v1.1e on Arduino UNO)
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [X] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [X] Windows
- [ ] Mac
- [ ] Linux

cheton, 2024-10-02:
The first issue seems to be related to the Electron app. Since it occurs infrequently and is difficult to reproduce, I currently don't have a solution for it.

For the 3D view with large files, it's better to slice them into smaller parts rather than rendering the entire object at once. This can help prevent freezing and improve performance. However, I currently don't have a straightforward approach to address this effectively. As a workaround, you can try disabling the 3D view when loading a large file to avoid performance issues.

closed 2024-10-02 by cheton

ColPaulR, 2024-10-03:
I can reproduce this one pretty reliably. What can I do to help pinpoint the problem?