cncjs/cncjs #865: General

AMunirudeen, 2024-06-14:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions
[image]

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

AMunirudeen, 2024-06-14:
[image]

AMunirudeen, 2024-06-14:
Write a g-code for machining the component attached on the CNC milling machine.

MitchBradley, 2024-06-14:
This is not a CNCjs software issue.  We do not provide gcode authoring services.

closed 2024-06-14 by MitchBradley