cncjs/cncjs #868: GRBL controller (FluidNC) is not getting status polling messages sent

ColPaulR, 2024-07-30:
#### Description

When I open COM7 and connect, I get a console prompt. However, the state of the GRBL controller is not updated. I have been manually commanding status updates via sending "?\n" on the console. When I do this a few times, the GRBL status window updates. In addition, the visualizer does not return after the last line of the G-code is sent and stay in a run state. Sending multiple "?\n" does not exit the G-code program even though an M2 was sent. Manually sending "M2" still does not switch the visualizer state to idle. I have to pause the visualizer, press the stop button and then "cycle start" button to return the visualizer to an idle/ready to run state.

How do I get CNCjs to start polling the status of the controller? Log file attached.
[electron.log](https://github.com/user-attachments/files/16421256/electron.log)

#### Versions

- CNCjs: 1.10.3
- Node.js: v22.5.1
- NPM: 10.8.2

#### How Do You Install CNCjs?

- [] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [X] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

ColPaulR, 2024-08-21:
As I have been studying the code, I think the issue here is the expectation to hit the CNCjs Reset button upon connection. If I read GrblController.js line 1267 correctly, this stops the workflow, resets the feeder, and issues a soft reset. Looking at http://wiki.fluidnc.com/en/support/gcode_senders#startup, this is not the desired flow. It would seem better to execute the stop workly and reset the feeder, then query the  controller for information immediately upon serial connection. Reading the [FluidNC documention](http://wiki.fluidnc.com/en/features/commands_and_settings#realtime-commands) and assuming the controller is not in motion, this may not lose much information. I get very concerned about losing my work position when hitting reset, but the soft reset appear to retain the $# information. 

I still need to confirm on my actual machine vs. development board that using the CNCjs Reset Button fixes the symptoms in this issue. Assuming so, could we add 
```
_this4.workflow.stop();
_this4.feeder.reset(); 
```

to the GrblController on serial connect function? What other information needs to be requested ($I, $G, etc.) if we are avoiding a soft reset?

ColPaulR, 2024-10-07:
This is not a DTR/RTS. This is CNCjs waiting for the restart string to be received before initiating autopolling of position and status. I REALY don't want to have to send a soft reset every time that I start up CNCjs. A soft reset is the only way currently to have FluidNC send the startup message. CNCjs really should be able to pick up operations from a fully initialized controller vs. forcing a soft reset. I've got a pull request in to FluidNC to command the startup message to resend, but this is a cumbersome way to start controlling FluidNC. 

Example would be executing laser operations in Lightburn and then switching to CNCjs for routing operations. I don't want to lose the co-registration that a controller reset may cause. FluidNC usually does not lose it's homing/location with a soft reset, but there are conditions where that could happen.

MitchBradley, 2024-10-08:
@cheton - It is great that you are working on CNCjs again.  That will help a lot of people.

I am the primary coder for FluidNC so I can assist if necessary.  Here are some tips to help you get started:
1. You don't need an actual CNC controller board.  Any bare ESP32 will do, as the default configuration is "test drive mode" that does not depend on connections to any motors or other I/O.  It must be an ESP32 WROOM - not WROVER and not -S2, -S3, -C3 or other variant.  (-S3 sort of works but is poorly supported, so use an ordinary one).
2. The easiest way to install the software is with https://installer.fluidnc.com/ .  It will connect to the ESP32 and let you install a release in a few clicks.  It also has tools for editing machine config files, but if you use a bare ESP32, you can just use the default test-drive mode and avoid making a config file.
3. The primary challenge in connecting to FluidNC is the fact that, unlike Arduinos, ESP32 boards do not automatically reset when the PC establishes a connection to the USB-serial chip.  This problem also occurs for GRBL boards that are connected via a direct UART RX/TX connection instead of via a USB-serial chip. 
4.  With GRBL, which only supports a single UART input channel, it is safe to just reset on every connection.  FluidNC supports multiple simultaneous connections from different devices - over the network, smart pendants on auxiliary UARTs, etc.  That means that, when CNCjs connects, it is possible that FluidNC is in the middle of running a job from another connection, so if you reset on connection, you could ruin that job.  So it is best to avoid resetting unless necessary, 
5. I will outline a reliable connection process below.

The procedure below essentially applies to direct-connect GRBL as well as FluidNC.

The first thing to do is to wait a couple of seconds to see if there is any output happening, in case the controller happens to be in the startup process.  During startup FluidNC sends information messages prefixed with [MSG: .  They typically come rapidly, but when it is trying to connect to a WiFi access point, there can be a couple of seconds delay between connection attempt messages.  If FluidNC is streaming messages like this, just wait for a Grbl ,.,. startup message.

If there is no output for a few seconds, you have to consider the possibility that GRBL/FluidNC happens to be in critical Alarm state, which can happen after a limit event.  In critical alarm state the only command it will respond to is ?.  All line-oriented commands are ignored.  So the first thing to try is sending a ?.  If there is no status report response to ?, the only thing you can do is send ^X and look for a Grbl startup message.

If you get a status report and the state is not Alarm, you have a good connection and you can issue various information queries like $I and $G to determine more of the status.

If you get a status report and the state is Alarm, try to unlock with $X.  If that does not work, send ^X to clear the critical alarm state, look for a Grbl message, then check the state again with ?.  If it is still Alarm, send $X to unlock.  At that point you are connected.

If you have any questions or issues, ask here, or if need be we can chat on Facebook or on the FluidNC Discord server, or any other channel you prefer.  When you get farther, I can help you trigger the critical alarm state so you can test that part of the connection process.

ColPaulR, 2024-10-08:
Thanks @MitchBradley for the response. If I read the code correctly, GrblController.js only executes initController() upon receiving the startup message from FluidNC, which as Mitch points out may not be available upon serial connection. There is a FluidNC pull request [#1314](https://github.com/bdring/FluidNC/pull/1314) that implements a reprint of the startup message when commanded ($GS). CNCjs works beautifully without a soft reset when the startup message is resent.

It would be great if the initController() function could be called in response to a "?" or "$I" command response as well as in response to the startup message and if !this.initialized (like in GrblController.js lines 584-589). This is the logic that Mitch describes above to determine if FluidNC is responding without doing the soft reset and waiting for the startup string to be received. I was assume that it would be up to the user to handle any alarm state once GrblController is actively updating the status with background "?" commands and parsed responses.

cheton, 2024-10-09:
Would it make sense to say that a practical approach is to send `?` for a status report (possibly multiple times) if the Grbl startup message isn't received within a few seconds after opening the connection?

MitchBradley, 2024-10-09:
> Would it make sense to say that a practical approach is to send ? for a status report (possibly multiple times) if the Grbl startup message isn't received within a few seconds after opening the connection?

That would work, with the following caveats:
1. If you are just looking for a "Grbl ..." message while discarding other messages like [MSG: ,...], the wait will need to be rather long, since WiFi connection can take 10 or more seconds in some cases.  If you pay attention to [MSG: ...], the wait can be optimized.  If there are no messages at all, there is no point in waiting more than 3 seconds.  But if there are {MSG: lines, you will need to wait longer.  The connection process prints up to 10 messages at intervals of 2 seconds before timing out.
2. It is not particularly likely that CNCjs will happen to connect just when FluidNC is starting, unless the user happens to synchronize them by turning on the FluidNC controller then immediately hitting Connect on CNCjs.  So, even though this situation must be handled, it is not the most likely.  Waiting 20+ seconds for a Grbl message when there are no [MSG lines will introduce an unnecessary and likely annoying delay in the usual case.
3. If there has been no output for more than 2 seconds, you can send a ? and you should receive a response very quickly, say less than 100 milliseconds.  There is no reason to try ? twice.  If you don't get a response to the first, you won't get a response to the second either.

cheton, 2024-10-09:
I created a PR (https://github.com/cncjs/cncjs/pull/880) to resolve this issue. I will attempt to run the test once I receive my ESP32 board. If you can modify the code on your side, feel free to share the results.

ColPaulR, 2024-10-09:
I'll give it a try tonight. Looks like just 1 file change in the codebase vs. a new branch?

ColPaulR, 2024-10-10:
I just tried this and it worked perfectly. I ran cnc -vvv. Attached is the console log.

[cncjs-868 log.txt](https://github.com/user-attachments/files/17320673/cncjs-868.log.txt)

closed 2024-10-15 by cheton

cheton, 2024-10-17:
This issue shall be resolved in [v1.10.4](https://github.com/cncjs/cncjs/releases/tag/v1.10.4)