cncjs/cncjs #871: Forbidden Access when using public IP

ScaredyCat, 2024-08-30:
#### Description

Desktop application will not start properly if using a public IP address. (for info I've replaced my actual ip with 99.99.99.99 

#### Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

When starting the desktop application on a mac using

`/Applications/CNCjs.app/Contents/MacOS/CNCjs --allow-remote-access --host 99.99.99.99 --port 62141
`
The application fails to start correctly, the result is a window with a white background and the text 

"Forbidden Access"

displayed. The startup log shows a warning:

`2024-08-30T15:50:43.232Z - warn app Error: Unauthorized IP address: ipaddr=99.99.99.99
`

#### Versions

- CNCjs: 1.10.3 Desktop, Mac

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [x] Mac
- [ ] Linux

#### Log

```

@plasma _PLAT % /Applications/CNCjs.app/Contents/MacOS/CNCjs --allow-remote-access --host 99.99.99.99 --port 62141
2024-08-30T15:44:35.083Z - info init Loading configuration from "/Users/andypowell/.cncrc"
(node:58957) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
(Use `CNCjs --trace-warnings ...` to show where the warning was created)
2024-08-30T15:44:35.202Z - info init Starting the server at http://99.99.99.99:62141
2024-08-30T15:44:35.277Z - warn app Error: Unauthorized IP address: ipaddr=99.99.99.99
    at /Applications/CNCjs.app/Contents/Resources/app/server/access-control.js:52:14
    at new Promise (<anonymous>)
    at authorizeIPAddress (/Applications/CNCjs.app/Contents/Resources/app/server/access-control.js:44:10)
    at _callee$ (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:116:58)
    at tryCatch (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:47:1357)
    at Generator.<anonymous> (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:47:4174)
    at Generator.next (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:47:2208)
    at asyncGeneratorStep (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:48:103)
    at _next (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:49:194)
    at /Applications/CNCjs.app/Contents/Resources/app/server/app.js:49:364
    at new Promise (<anonymous>)
    at /Applications/CNCjs.app/Contents/Resources/app/server/app.js:49:97
    at /Applications/CNCjs.app/Contents/Resources/app/server/app.js:135:19
    at Layer.handle [as handle_request] (/Applications/CNCjs.app/Contents/Resources/app/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/Applications/CNCjs.app/Contents/Resources/app/node_modules/express/lib/router/index.js:317:13)
    at /Applications/CNCjs.app/Contents/Resources/app/node_modules/express/lib/router/index.js:284:7
2024-08-30T15:45:50.434Z - warn app Error: Unauthorized IP address: ipaddr=99.99.99.99
    at /Applications/CNCjs.app/Contents/Resources/app/server/access-control.js:52:14
    at new Promise (<anonymous>)
    at authorizeIPAddress (/Applications/CNCjs.app/Contents/Resources/app/server/access-control.js:44:10)
    at _callee$ (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:116:58)
    at tryCatch (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:47:1357)
    at Generator.<anonymous> (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:47:4174)
    at Generator.next (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:47:2208)
    at asyncGeneratorStep (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:48:103)
    at _next (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:49:194)
    at /Applications/CNCjs.app/Contents/Resources/app/server/app.js:49:364
    at new Promise (<anonymous>)
    at /Applications/CNCjs.app/Contents/Resources/app/server/app.js:49:97
    at /Applications/CNCjs.app/Contents/Resources/app/server/app.js:135:19
    at Layer.handle [as handle_request] (/Applications/CNCjs.app/Contents/Resources/app/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/Applications/CNCjs.app/Contents/Resources/app/node_modules/express/lib/router/index.js:317:13)
    at /Applications/CNCjs.app/Contents/Resources/app/node_modules/express/lib/router/index.js:284:7
2024-08-30T15:45:50.642Z - warn app Error: Unauthorized IP address: ipaddr=99.99.99.99
    at /Applications/CNCjs.app/Contents/Resources/app/server/access-control.js:52:14
    at new Promise (<anonymous>)
    at authorizeIPAddress (/Applications/CNCjs.app/Contents/Resources/app/server/access-control.js:44:10)
    at _callee$ (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:116:58)
    at tryCatch (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:47:1357)
    at Generator.<anonymous> (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:47:4174)
    at Generator.next (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:47:2208)
    at asyncGeneratorStep (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:48:103)
    at _next (/Applications/CNCjs.app/Contents/Resources/app/server/app.js:49:194)
    at /Applications/CNCjs.app/Contents/Resources/app/server/app.js:49:364
    at new Promise (<anonymous>)
    at /Applications/CNCjs.app/Contents/Resources/app/server/app.js:49:97
    at /Applications/CNCjs.app/Contents/Res
![screenshot](https://github.com/user-attachments/assets/9e81ba21-18b4-4d76-9e11-06635c7b04e1)
![screenshot](https://github.com/user-attachments/assets/7d7584b6-6c0e-4f89-9c8c-fb0c6b7e6715)
ources/app/server/app.js:135:19
    at Layer.handle [as handle_request] (/Applications/CNCjs.app/Contents/Resources/app/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/Applications/CNCjs.app/Contents/Resources/app/node_modules/express/lib/router/index.js:317:13)
    at /Applications/CNCjs.app/Contents/Resources/app/node_modules/express/lib/router/index.js:284:7

```

ScaredyCat, 2024-08-30:
[image]

cheton, 2024-10-02:
Public IP addresses are prohibited due to security concerns. You can either modify the code to bypass public IP addresses or set up a reverse proxy to route through the internal network.

closed 2024-10-02 by cheton