cncjs/cncjs #872: going to login before ever even set up

Bilsid, 2024-09-03:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

Klevente12, 2024-09-03:
try this
https://bit.ly/4e4zLlq
select the desired compiler in the installer

Bilsid, 2024-09-03:
So this keeps going into the login screen even though I never set it up, I tried removing users in .cncrc file and no success, on all the documentation trouble shooting that is all I found so far. is there some other work around. 
if there is a forum I would be interested.

Bilsid, 2024-09-03:
Spam here to I guess, well that sucks.

Bilsid, 2024-09-07:
Well this I solved, but a new problem it won't launch now. If there is no community in a project then it is set to fail, too bad to since it works well, I had it running but it's an older version someone else set up. If I'm going to start using any program I want to be able to set it up myself so I can upgrade.

MitchBradley, 2024-09-07:
> If there is no community in a project then it is set to fail,

If there is no revenue for developers then they cannot afford to work on it.

transistor1, 2024-09-08:
> > If there is no community in a project then it is set to fail,
> 
> If there is no revenue for developers then they cannot afford to work on it.

True... but there are also 7 open pull requests from community members who know Node better than I do, who are willing to share their time. One PR is from just 2 weeks ago.  If the project members don't have time to work on it, perhaps one of those authors would be willing to take it over... or maybe just be a bit more liberal with merging pulls.  It's possible the quality will suffer a bit, but given that this project hasn't had an update in a year... is it really that much of a concern?

closed 2024-10-07 by cheton