cncjs/cncjs #874: Help Needed with ShuttleXpress on Raspberry Pi 5 for CNCjs

mgcdunbar, 2024-09-12:
#### Description

I'm looking for help getting a ShuttleXpress working on a Raspberry Pi 5 to control jogging in CNCjs. I’ve explored existing projects such as [Ben Suffolk's CNCjs Pendant Shuttle](https://github.com/bensuffolk/cncjs-pendant-shuttle), but these rely on older versions of CNCjs with the cancelJog PR #512 applied. Unfortunately, the dependencies in these projects have either undergone breaking changes or have been superseded.

I don't have the experience necessary to adapt the code from these earlier projects to work with the current version of CNCjs. I'd greatly appreciate any guidance or help in implementing an up-to-date solution that supports both CNCjs and the ShuttleXpress.

Thank you in advance for any help you can offer.

#### Versions

- CNCjs: 1.10.3
- Node.js: 20.17.0
- NPM: 10.8.3

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

cheton, 2024-10-02:
PR #512 was too much complex for managing the feeder queue. Instead, I will simplify it by adding a straightforward `jogCancel` command that directly sends the `0x85` character to Grbl.

cheton, 2024-10-02:
This PR has been merged to the master branch.

closed 2024-10-02 by cheton