cncjs/cncjs #879: Unlock / xpro v5 becomes unresponsive after upgrading from 1.9.22 to 1.10.3

jason-caf, 2024-10-04:
#### Description

After upgrading the console would lock up after connecting; unlock command and then it seemed to hang.  Power cycled controller and downgrade to 1.9 = fixed.

Rolling back to 1.9.22 fixed the issue.

#### Versions

- CNCjs: 1.10.3

#### How Do You Install CNCjs?

- [] NPM
- [x ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

####Console info (on working version)

NCjs 1.9.22 [Grbl]
Connected to COM7 with a baud rate of 115200
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20220621]
[MSG:Compiled with ESP32 SDK:v3.2.3]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Door switch on pin GPIO(16)]
[MSG:Macro Pin 1 GPIO(13)]
[MSG:Macro Pin 2 GPIO(0)]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,1300.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,2820.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,2820.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-110.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:Relay spindle Output:GPIO(25), Enbl:GPIO(4), Dir:None]
[MSG:Client Started]
[MSG:Connecting jlan6]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connected with 192.168.50.10]
[MSG:Start mDNS with hostname:http://CNC_xPRO_V5.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
client> $$
[MSG:'$H'|'$X' to unlock]
$0=5 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=100.000 (Homing locate feed rate, mm/min)
$25=1000.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=10.000 (Homing switch pull-off distance, millimeters)
$30=23000.000 (Maximum spindle speed, RPM)
$31=24000.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=200.000 (X-axis travel resolution, step/mm)
$101=31.830 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$103=31.830
$104=100.000
$105=100.000
$110=3000.000 (X-axis maximum rate, mm/min)
$111=6000.000 (Y-axis maximum rate, mm/min)
$112=2500.000 (Z-axis maximum rate, mm/min)
$113=6000.000
$114=1000.000
$115=1000.000
$120=100.000 (X-axis acceleration, mm/sec^2)
$121=100.000 (Y-axis acceleration, mm/sec^2)
$122=100.000 (Z-axis acceleration, mm/sec^2)
$123=100.000
$124=200.000
$125=200.000
$130=1300.000 (X-axis maximum travel, millimeters)
$131=2820.000 (Y-axis maximum travel, millimeters)
$132=110.000 (Z-axis maximum travel, millimeters)
$133=2820.000
$134=300.000
$135=300.000
ok
client> $X
[MSG:Caution: Unlocked]
ok
feeder> G91
feeder> G0 Y5
feeder> G90
ok
ok
ok
>

cheton, 2024-10-07:
@jason-caf 

Similiar to https://github.com/bdring/Grbl_Esp32/issues/975

You might need to perform a soft reset to reset your ESP32.

Have you ever tried adjusting the `Set DTR line status upon opening` or `Set RTS line status upon opening` options on CNCjs 1.10.3 by setting them to either `SET` or `CLR`?

[image]

cheton, 2024-10-07:
For your reference:
http://wiki.fluidnc.com/en/support/gcode_senders

A reset using RTS/DTR may be necessary:
[image]

cheton, 2024-10-17:
This should be resolved in 1.10.4

closed 2024-10-17 by cheton