cncjs/cncjs #889: GRBL doesn't

lgojniczek, 2024-10-25:
#### Description

New version does not open serial communication with Grbl, I think it has something in common with last changes in UART communication (without resetting the connection upon opening). When I initialize the connection with (for example) Picocom first then the new version of cncjs works as expected thus it provides me to the thinking of connection reset prior to use it properly.

How can i avoid this?

#### Versions

- CNCjs: 1.10.4
- Node.js: v18.19.0
- NPM: 9.2.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl-Mega
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

cheton, 2024-10-28:
Hi @lgojniczek,

Could you try running `./bin/cncjs -vvv` to collect the logs? Also, if possible, could you test with another Arduino board, like the Arduino Uno or Arduino Nano with Grbl? I'm curious if this issue is specific to [grbl-Mega](https://github.com/gnea/grbl-Mega).

lgojniczek, 2024-10-28:
Hi cheton,

I have tested it on just pure Grbl 1.1h on Arduino Uno and it works like a charm. So this must be something with grbl-Mega. Apparently I have noticed that if I choose **Marlin** in the Connection widget it's all good and connection is established.

cheton, 2024-10-30:
I recalled an edge case, with certain Grbl builds, the Grbl controller will not respond if the sender continuous sends data (e.g., status query `?`) before establishing a serial connection. To address this, a new option might be needed to maintain the original behavior by waiting to send status queries until after receiving the Grbl startup string.

cheton, 2024-10-31:
@lgojniczek Locate the line with `queryActvity()` in `GrblController.js` and remove it as shown below:

[image]

I tested this change earlier, and it should resolve the issue with `grbl-Mega`.

cheton, 2024-10-31:
I will add a new option to distinguish between USB serial communication and UART communication for the Grbl controller.

cheton, 2024-10-31:
It will be fixed in PR #893. A waiting time of 1 second should be enough for grbl-Mega.

lgojniczek, 2024-10-31:
Great, thank you.

closed 2024-10-31 by cheton

cheton, 2024-10-31:
@lgojniczek It's fixed in CNCjs [v1.10.5](https://github.com/cncjs/cncjs/releases/tag/v1.10.5)

lgojniczek, 2024-10-31:
Many thanks cheton, I really appreciate that.