cncjs/cncjs #890: Need Newest version CncJs Without the M6 add-in

NabCube, 2024-10-29:
#### Description

I am using a CNC with a Auto Tool Changer which uses the M6 Tool command in the gcode. Its Version 1.9.8win64 and does NOT have the newer implement of the M6 change for manual tool changes which Allows my ATC to work correctly...

--Question: Is there a way to have a Fork or option of the newer CncJs program without the semi-auto or manual tool change with the m6 command like in the newest versions past 1.9.8??? 

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [X] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ X] Windows
- [ ] Mac
- [ ] Linux

cheton, 2024-10-30:
You have to upgrade CNCjs to at least 1.9.26 or higher to enable support for the M6 command.

cheton, 2024-10-30:
Let's rephrase your question: You're requesting a new option to disable the M6 tool change notification for manual tool changes.

In the meantime, you can modify your code by removing the code block that processing the M6 command. Search for the "M6 Tool Change" keyword in `server/controllers/Grbl/GrblController.js`:

```js
// M6 Tool Change
if (_.includes(words, 'M6')) {
    log.debug('M6 Tool Change');
    this.feeder.hold({ data: 'M6', msg: originalLine }); // Hold reason

    // Surround M6 with parentheses to ignore
    // unsupported command error. If we nuke the whole
    // line, then we'll likely lose other commands that
    // share the line, like a T~.  This makes tool
    // changes complicated.
    line = line.replace('M6', '(M6)');
}
```

NabCube, 2024-10-30:
Thank.. I tried it but with no change..maybe i edited it wrong...I want to
use CncJs for the Adaptive Tool length Offsets with my machine. I run
FluidNC with grbl.. I like CNCjs and want it as my main App. So whatever
will allow the M61 T# command (to set tool #) and M6 Command (To change
tool) to work correctly with the newest version would be fantastic... I
have a Rapid Auto Tool Changer, Automated probe arm and automated dust
boot.. This would be the perfect control software for me and others using
the ATC i use..

NabCube, 2024-10-30:
I edited it again, must have miss something or added space. I got it working with the newest 1.10.4.. This is awesome... Thank u again..

closed 2025-01-14 by cheton