cncjs/cncjs #894: Developer: cncjs version 1.10.5 build with yarn fails - problem with jest

atmelino, 2024-11-07:
#### Description
OS: Ubuntu 24.04
The exact same procedure worked with cncjs version 1.10.3 without problems

Procedure:

Install Node version manager:
curl -o- https://raw.githubusercontent.com/nvm-sh/nvm/v0.40.1/install.sh | bash

Install Node version 14:
nvm install 14

Set Node version to be used:
nvm use 14
Output:
Now using node v14.21.3 (npm v6.14.18)

Install yarn:
npm install -g yarn

Download/Clone cncjs code: 
git clone https://github.com/cncjs/cncjs.git

Change into code directory:
cd cncjs

Install the dependencies:
yarn

Build the app:
yarn build
Leads to error:
ERROR in [eslint] Failed to load plugin 'jest' declared in '.eslintrc.js': Unexpected token '||='

Start the cncjs server:
bin/cncjs

Open web browser at 
http://127.0.1.1:8000/
Leads to error:
Error: Failed to lookup view "index.hbs" in views directories "github/cncjs/dist/cncjs/app" or "github/cncjs/dist/cncjs/server/views"

web browser displays:
An error occurred!
Error:

#### Versions

- CNCjs: 1.10.5
- Node.js: v14.21.3
- NPM: v6.14.18

#### How Do You Install CNCjs?

- [ ] NPM
- [ ] Download the CNCjs Desktop Application
- [ ] git clone source for development
#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

cheton, 2024-11-11:
Node.js v18 is required starting from CNCjs 1.10.4. You can check the release notes here:

https://github.com/cncjs/cncjs/releases/tag/v1.10.4

> This release now supports Node.js v18 and newer versions. If you experience authentication issues, upgrading your Node.js version to v18 or higher may help resolve any runtime problems.

closed 2024-11-11 by cheton

atmelino, 2024-11-13:
Confirmed: 
cncjs version 1.10.5 build with Node.js v 18 works.

Install Node version 18:
nvm install 18

Set Node version to be used:
nvm use 18

Would it make sense to update the README.md  at https://github.com/cncjs/cncjs to require Node.js version 18 or higher?

Currently:

## Supported Node.js Versions

| Version | Supported Level |
| :------ |:--------------- |
| <= 10   | Not supported   |
| 12      | Supported       |
| >= 14   | Recommended     |