cncjs/cncjs #904: Complete manual control over tool:change

Jnesselr, 2025-02-05:
I noticed there's a `tool:change` macro which works based on the tool change policy. However, even the `TOOL_CHANGE_POLICY_MANUAL_TOOL_CHANGE_CUSTOM` policy only allows so much control over everything. I would like complete control over all of the g-code that is returned here. That doesn't currently seem to be possible, but I'm just wondering if I missed something. If not, what's the best way to go about adding this feature?

cheton, 2025-02-10:
@Jnesselr I'm aware of this issue, which is why I haven't released a new version yet. I'll create another PR with this option. Thank you for confirming.

Jnesselr, 2025-03-06:
I've been following along with the PR, but wanted to ask for clarification on something. It doesn't look like you're adding a command to give full G-Code control. All of the commands either send M6, ignore M6, or call `tool:change`. Inside `tool:change`, `TOOL_CHANGE_POLICY_MANUAL_TOOL_CHANGE_CUSTOM_PROBING` exists to allow you to insert G-Code lines just for the probing part, but there doesn't look like there's anything that allows full control over what `tool:change` is doing.

Mostly what I'm running into is that I don't need a move back to the tool change position after running G-Code, or a pause to restart the spindle, and I don't want it to move back to its previous location with the spindle modal change. The G-Code in my file stops the spindle itself with an `M5` , and I don't really need to set `G90` either.

I'm mostly asking for clarification because I would hate to have you do a bunch of work and then realize I had miscommunicated something.