cncjs/cncjs #921: My ultimate bee stops randomly during the cutting job

fabioputz, 2025-06-18:
Hello everyone, I'm new to Github and I hope you guys can help me.

I'm having troubles with my Ultimate Bee CNC machine run with CNCjs.
When I'm running a milling job (sometimes quite heavy, 3-4Mb g-code) the machine stops randomly, in different stops. Sometime it gives me an 'Error:2', sometimes it doesn't. When it gives me the error message I can manage to continue the job pushing on the 'play' button of CNCjs, when no error message is shown there is nothing I can do to save my milling job.
- My controller is a XPro V5
- I use Fusion360 to create the G-code with official GRBL post processor from Autodesk
- I use Fusion360 on Mac
- The G-code apparently has no error or bad coding

The strange thing is that I started having this issue only recently. I have been using my machine for many months with no problem but since two days ago I'm facing this issue. I change nothing in the setup hardware/software.

I have read online that it might be a communication issue due to bad data transfer (I use a 4 meters long USB cable). Do you think that can be the issue? Do you think shielding somehow the cable might solve the issue?

Any suggestion is super welcome!!!!

MitchBradley, 2025-06-18:
Error 2 is "bad number format".  It typically means that characters are missing from the GCode line, so, yes, communications problems are a likely cause.

XProV5 runs either Grbl_ESP32 or its successor FluidNC.  They support running jobs from the controller's SD Card, so you could try uploading your gcode to it using WebUI then running with $sd/run=<filename> (from cncjs) or hitting the run button with WebUI.  SD Card runs tend to be immune to serial line problems.

fabioputz, 2025-06-19:
Unfortunately still nothing positive :-(

[image]

MitchBradley, 2025-06-19:
We really should do this on the FluidNC Discord.  This is not a CNCjs issue.

fabioputz, 2025-06-19:
> We really should do this on the FluidNC Discord. This is not a CNCjs issue.

Right! I just subscribed on Discord

MitchBradley, 2025-06-19:
Create a thread in #help-forum.  A good title would be "XProV5 Installation".

closed 2025-06-19 by MitchBradley